cabbagecreek/Marlin3DprinterTool #13: 3 Point Bed Leveling Option

technoplunk, 2016-09-27:
I notice on the Bed Level tab that it appears only 4 point bed leveling is supported. Some people use 3 point leveling. It should be a bit simpler to implement since a plane is easiest to define with only 3 points. 

My bed only uses 3 points so I could test this if implemented.

cabbagecreek, 2016-09-27:
Do you know if the singlepoint is always on the right side? If so the GUI
part is easy.

The calculation is totaly different with 3 points instead of 4 points. The
surface image must atleast have 4 points.
The surfacescan and the bedlevel is done with the same code.
I think I have to create a separate function for 3 point bedlevel.

I will do a try on a 3 part bedlevel.

Many regards

technoplunk, 2016-09-27:
Usually the single point is in the front. With the other two in the back left and back right. 

Specifically my case I'm using an MK3 heated bed which is commonly mounted on the 4 corners but there is an additional hole on the front near the electrical connectors which I'm using for the third point rather than the two front corners.

Another thing to keep in mind in the future is that there are a few other shapes of heatbeds out there, I've seen both round and hexagonal that are more commonly used with delta printers. I don't have one of those but it might be worth keeping in mind when designing that screen that other requests may come in the future.

Thanks again for looking at this, I look forward to doing more testing.

cabbagecreek, 2016-09-28:
GUI part Done. 
Will do the calculation and live test tonight. 

Skickat från min iPhone

cabbagecreek, 2016-09-28:
Thanks for this information.

I have done singlepoint left and singlepoint right, BUT not singlepoint
front.. I will add that..

The calculation should be redone anyway, so this is a small change.

I think I stop at the squered and triangular shaped adjustment modells.
I dont have enought expiriance of delta printers to do bedleveling for them.
I belive that the delta printers have a fixed bedsurface and use the three
enstops to adjust the "corners".
This is complicated and many delta printers have a z-probe to do ABL or
MBL.
The buildsurface is smaller on delta printers and the lack of movement
makes it easier to create a flat buildsurface.

Maybe a 4:th 3D printer will be a Delta ;-)

Have a wounderful day

cabbagecreek, 2016-10-04:
Included in the v0.4-beta Release

closed 2016-10-04 by cabbagecreek

cabbagecreek, 2016-12-02:
Skickat från min iPhone