cabbagecreek/Marlin3DprinterTool #17: Config editor/Migration fails without admin rights

derpicknicker1, 2016-10-04:
I have to start the software as administrator to use it as desired. When i try to set the paths for migration i get an error (by the way...you released the debug version 😄 ). 
Something like "Access forbidden for 'c:\Program Files (x86)\Marlin3Dprinter Tool\MarlinEditorConfiguration.xml'" (for me its in german).
Same notification now also on startup after updating to newest release.

Using up to date Win10.

derpicknicker1, 2016-10-05:
Maybe you should use something like `Environment.SpecialFolder.MyDocuments` or `Environment.SpecialFolder.LocalApplicationData` as path for saving files. I think this is a problem of writing to the installation folder of the programm.

[More Informations on MSDN](https://msdn.microsoft.com/en-us/library/system.environment.specialfolder(v=vs.110).aspx)

cabbagecreek, 2016-10-05:
I will make an issue of this and try to correct this soon.

It is not my intension to do release in debug mode. I am a newbie on Git
and Gitflow but I try to be better. :-)

MANY REGARDS!!!!

derpicknicker1, 2016-10-05:
The debug mode has nothing to do with github. You have to set this in VisualStudio build options(?)  (Debug/Release).

cabbagecreek, 2016-10-05:
I have started the WiX installer to move the configuration files to
ProgramData. I have some time this evening to test this.

cabbagecreek, 2016-10-05:
I have moved the config files to ProgramData

closed 2016-10-05 by cabbagecreek