cabbagecreek/Marlin3DprinterTool #18: 250,000 Baud

derpicknicker1, 2016-10-04:
Why there is no option for 250,000 Baud in the dropdown field for baudrate selection? There ia a option for 25**6**,000 Baud isntead. Maybe a typo? Because this option works for me witn my board configured to 250,000 Baud.

cabbagecreek, 2016-10-05:
It is actually not a typo. The 250 000 Baud is used with arduino but there is OS and languages that dont support this baudrate. The 250000 is a non-standard value.
( http://forums.reprap.org/read.php?1,163707 )
( https://github.com/foosel/OctoPrint/issues/222 )
( https://github.com/MarlinFirmware/Marlin/issues/205 )

The documentation to the library used (ZylSerial) dont have 250000 without using the Custom baudrate = 0 and force it to 250000. 
( http://www.zylsoft.com/zylserialportnet/html/3bf56933-81a1-a222-58f2-229a17381927.htm )

Serial is sometimes indulgence and tries to understand even if the baudrate is wrong. Using 256000 and have 25000 in Marlin Firmware is to use this fault tollerant feauture. 

I dont belive there is a disadvantage to use the lower 230400 Baudrate. 

Is it possible that you can change the Marlin Configuration.h to use  230400 instead of 250000?
I tested and 230400 is the higest baudrate I got working an a MKS 1.3 board. 
A more secure value is to use 115200. The printer is so slow that the transfer at 115200 will transfer GCODE faster than the printer is able to execute the GCODE. 

configuration.h
...
// SERIAL_PORT selects which serial port should be used for communication with the host.
// This allows the connection of wireless adapters (for instance) to non-default port pins.
// Serial port 0 is still used by the Arduino bootloader regardless of this setting.
// :[0,1,2,3,4,5,6,7]
#define SERIAL_PORT 0

// This determines the communication speed of the printer
// :[2400,9600,19200,38400,57600,115200,250000]
**#define BAUDRATE 230400**

// Enable the Bluetooth serial interface on AT90USB devices
//#define BLUETOOTH
...

cabbagecreek, 2016-10-05:
The 250000 baudrate is added even if its not a standard baudrate. The 250000 baudrate is optimized for the Arduino Mega and many printers use the 250000 baudrate as default.

closed 2016-10-05 by cabbagecreek