cabbagecreek/Marlin3DprinterTool #41: Tool does not start anymore, STL Preview does not work.

derpicknicker1, 2016-10-27:
I uninstalled 0.8 beta and installed 0.9 beta. Now i am not able to start the program anymore. i double click te desktop shortcut but nothing happens. Also i get no stl file preview in the file explorer and i don't understand what you mean when you say "disk clean" in release notes.
MarlinEditor does start from startmenu. MarlinSTLviewer too. But not the Marlin 3DprinterTool.

derpicknicker1, 2016-10-27:
https://gist.github.com/derpicknicker1/06918f357164ac7cc671c6abd613e573

cabbagecreek, 2016-10-27:
I am working on this.. new msi is under upload. A new STL viewer and STL thumbnail debug document is included in the release

derpicknicker1, 2016-10-27:
Thanks for your fast reply. I managed to see file thumbnails. And the File preview now also shows the thumbnail now. I'm not sure if i should be able to rotate the model in preview (i can't do that).
Hope the firmwareTool will also soon work again.

cabbagecreek, 2016-10-28:
I focused on the Marlin3dprinterTool not the Marlin Editor..
Please descride whats wrong with the FirmwareTool.. Is it connection to Arduino IDE of is the MarlinEditor not working

I gave been soooo focused on the Shell Extension that I may have missed some other part. 
If it is possible to clean the windows log and restart the MarlinEditor to see if the Windows log capture the error.

MANY regards

derpicknicker1, 2016-10-28:
The MarlinEditor (edit config.h) works fine. But the main tool (Marlin3dprinterTool ) does not start. I get a "File not found"-Exception. There are not really more infos in the log than the 2 links above provide. Maybe this helps (it's german, sorry):
```
Anwendung: Marlin3DprinterTool.exe Frameworkversion: v4.0.30319 Beschreibung: Der Prozess wurde 
aufgrund einer unbehandelten Ausnahme beendet. Ausnahmeinformationen: 
System.IO.FileNotFoundException bei Nevron.Chart.View.NView.CreateContext() bei 
Nevron.Chart.WinForm.NControlView.CreateContext() bei 
Nevron.Chart.View.NView.Init(Nevron.Chart.NDocument, Nevron.GraphicsCore.NSize, 
Nevron.GraphicsCore.NResolution) bei 
Nevron.Chart.WinForm.NControlView..ctor(Nevron.Chart.NDocument, System.Windows.Forms.Control) bei
 Nevron.Chart.WinForm.NChartControl..ctor() bei 
Marlin3DprinterTool.FrmMarlin3DprinterTool.InitializeComponent() bei 
Marlin3DprinterTool.FrmMarlin3DprinterTool..ctor() bei Marlin3DprinterTool.Program.Main()

```

cabbagecreek, 2016-10-28:
Look at the troubleshoot guide. Its for the STL viewer / thumbnail but includes the total package.

1. Download the MSI.. It is updated and the rtf document (Troubleshootguide for STL viewer/thumbnail)
2. Save the files inside "C:\ProgramData\cabbagecreek\Marlin3DprinterTool"
3. Uninstall
4. Remove the directory "C:\Program Files (x86)\Marlin3DprinterTool" 
5. Remove the directory "C:\ProgramData\cabbagecreek\Marlin3DprinterTool"
6. Install
7. Replace the files inside "C:\ProgramData\cabbagecreek\Marlin3DprinterTool" with the saved ones.
8. Follow the rtf document. First part shows what should be installed. Are there missing files?
9, Secund part helps you get the STL viewer/STL thumbnail working

derpicknicker1, 2016-10-28:
I did the 9 steps. There are no missing files, but the problem still exists.

cabbagecreek, 2016-10-28:
Can I help you thru skyp or Teamviewer later today?

cabbagecreek, 2016-10-30:
Have you tested v0.10-beta?? I want to find the reason why the Marlin3DprinterTool dont start

derpicknicker1, 2016-10-31:
Sorry. There was no time to read along here on the weekend. I will test today.

cabbagecreek, 2016-11-08:
I may have found the problem.. contact me at c a b b a g e c r e e k @ g m a i l , c o m 
and I will send you a new msi file

cabbagecreek, 2016-11-10:
Issue should be fixed from v0.11-beta

Uninstall previous Installation and install v0.11-eta or higher

closed 2016-11-10 by cabbagecreek

derpicknicker1, 2016-11-11:
Yes. 0.11 fixed it. I'm sorry that i dont have replied, but i was very busy the last days.
Thank you!

cabbagecreek, 2016-11-15:
The next upcomming v0.12-beta will have some more features.. Try to do the
update before this weekend

Regards