cabbagecreek/Marlin3DprinterTool #45: [Feature] Tab for EEPROM settings

derpicknicker1, 2016-11-11:
Hey. Thank you so far for this great tool and the ongoing development.
A thing i miss within the tool is a Tab which shows my EEPROM settings (as shown in the dialog on connection). I think it will be very usefult to have a tab for them to view and - much more impartant - to edit them. 
For example when i tune a new printer, i want to be able to edit "steps per mm" and things like that.
Would be very nice, if you will find some time to implement this.

cabbagecreek, 2016-11-12:
This is a VERY good idea...  There is some loose threads concerning EEPROM and update. To have all that in run-time and able to change value first in EEPROM and then transfer to current Firmware file(s)

I am working now on the Automatic MBL.. and this will be of enormus help.. So I think this feature will be made soon

cabbagecreek, 2016-11-17:
Do you want the next v0.12-beta and test it.. It contains some of the data from the EEPROM..
I want feedback if there is something you are missing

mail me at c a b b a g e c r e e k @ g m a i l . c o m

Pipshag, 2016-11-19:
EEPROM tab is working correctly and has some nice formatting to it.

cabbagecreek, 2016-11-20:
I find this issue fixed

closed 2016-11-20 by cabbagecreek