cabbagecreek/Marlin3DprinterTool #48: Default PID values are duplicated in the dropdown list

psavva, 2016-11-17:
I can see the values for
DEFAULT_Kp
DEFAULT_Ki
DEFAULT_Kd
and 
DEFAULT_bedKp
DEFAULT_bedKi
DEFAULT_bedKd

are duplicated 3 times in the dropdown

cabbagecreek, 2016-11-17:
What version of Marlin

Send a screenshoot

psavva, 2016-11-17:
[image]

I'm using RC7

cabbagecreek, 2016-11-17:
This looks like an older version of M3DPT..

cabbagecreek, 2016-11-18:
Uninstall and DELETE the files in C:\programfilesx86\Marlin3DprinterTool.  <== IMPORTANT!!

Then reinstall with the MSI you got in mail. The MSI you got in mail is the v0.13-beta 

Check if the problem still occur and make a note if it does

cabbagecreek, 2016-11-28:
This is an old version. Fixed in the current version

closed 2016-11-28 by cabbagecreek