cabbagecreek/Marlin3DprinterTool #52: Clicking "Copy to Clipboard" raises an Exception when nothing is selected

psavva, 2016-11-17:
See the end of this message for details on invoking 
just-in-time (JIT) debugging instead of this dialog box.

************** Exception Text **************
System.ArgumentNullException: Value cannot be null.
Parameter name: text
   at System.Windows.Forms.Clipboard.SetText(String text, TextDataFormat format)
   at System.Windows.Forms.Clipboard.SetText(String text)
   at MarlinComunicationHelper.FrmFirmware.btnExtruderFirmwareCopyToClipboard_Click(Object sender, EventArgs e)
   at System.Windows.Forms.Control.OnClick(EventArgs e)
   at System.Windows.Forms.Button.OnClick(EventArgs e)
   at System.Windows.Forms.Button.OnMouseUp(MouseEventArgs mevent)
   at System.Windows.Forms.Control.WmMouseUp(Message& m, MouseButtons button, Int32 clicks)
   at System.Windows.Forms.Control.WndProc(Message& m)
   at System.Windows.Forms.ButtonBase.WndProc(Message& m)
   at System.Windows.Forms.Button.WndProc(Message& m)
   at System.Windows.Forms.Control.ControlNativeWindow.OnMessage(Message& m)
   at System.Windows.Forms.Control.ControlNativeWindow.WndProc(Message& m)
   at System.Windows.Forms.NativeWindow.Callback(IntPtr hWnd, Int32 msg, IntPtr wparam, IntPtr lparam)

************** Loaded Assemblies **************
mscorlib
    Assembly Version: 4.0.0.0
    Win32 Version: 4.6.1055.0 built by: NETFXREL2
    CodeBase: file:///C:/Windows/Microsoft.NET/Framework/v4.0.30319/mscorlib.dll
----------------------------------------
Marlin3DprinterTool
    Assembly Version: 1.0.0.0
    Win32 Version: 1.0.0.0
    CodeBase: file:///C:/Program%20Files%20(x86)/Marlin3DprinterTool/Marlin3DprinterTool.exe
----------------------------------------
System.Windows.Forms
    Assembly Version: 4.0.0.0
    Win32 Version: 4.6.1055.0 built by: NETFXREL2
    CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System.Windows.Forms/v4.0_4.0.0.0__b77a5c561934e089/System.Windows.Forms.dll
----------------------------------------
System
    Assembly Version: 4.0.0.0
    Win32 Version: 4.6.1075.0 built by: NETFXREL3STAGE
    CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System/v4.0_4.0.0.0__b77a5c561934e089/System.dll
----------------------------------------
System.Drawing
    Assembly Version: 4.0.0.0
    Win32 Version: 4.6.1068.2 built by: NETFXREL3STAGE
    CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System.Drawing/v4.0_4.0.0.0__b03f5f7f11d50a3a/System.Drawing.dll
----------------------------------------
MarlinComunicationHelper
    Assembly Version: 1.0.0.0
    Win32 Version: 1.0.0.0
    CodeBase: file:///C:/Program%20Files%20(x86)/Marlin3DprinterTool/MarlinComunicationHelper.DLL
----------------------------------------
ZylSerialPort
    Assembly Version: 1.54.5614.24643
    Win32 Version: 1.54.5614.24643
    CodeBase: file:///C:/Program%20Files%20(x86)/Marlin3DprinterTool/ZylSerialPort.DLL
----------------------------------------
System.Windows.Forms.DataVisualization
    Assembly Version: 4.0.0.0
    Win32 Version: 4.0.30319.33440
    CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System.Windows.Forms.DataVisualization/v4.0_4.0.0.0__31bf3856ad364e35/System.Windows.Forms.DataVisualization.dll
----------------------------------------
FastColoredTextBox
    Assembly Version: 2.16.11.0
    Win32 Version: 2.16.11.0
    CodeBase: file:///C:/Program%20Files%20(x86)/Marlin3DprinterTool/FastColoredTextBox.DLL
----------------------------------------
Nevron.Chart.WinForm
    Assembly Version: 16.5.16.12
    Win32 Version: 16.5.16.12
    CodeBase: file:///C:/Program%20Files%20(x86)/Marlin3DprinterTool/Nevron.Chart.WinForm.DLL
----------------------------------------
Nevron.Chart
    Assembly Version: 16.5.16.12
    Win32 Version: 16.5.16.12
    CodeBase: file:///C:/Program%20Files%20(x86)/Marlin3DprinterTool/Nevron.Chart.DLL
----------------------------------------
Nevron.System
    Assembly Version: 16.5.16.12
    Win32 Version: 16.5.16.12
    CodeBase: file:///C:/Program%20Files%20(x86)/Marlin3DprinterTool/Nevron.System.DLL
----------------------------------------
Nevron.Presentation
    Assembly Version: 16.5.16.12
    Win32 Version: 16.5.16.12
    CodeBase: file:///C:/Program%20Files%20(x86)/Marlin3DprinterTool/Nevron.Presentation.DLL
----------------------------------------
System.Configuration
    Assembly Version: 4.0.0.0
    Win32 Version: 4.6.1055.0 built by: NETFXREL2
    CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System.Configuration/v4.0_4.0.0.0__b03f5f7f11d50a3a/System.Configuration.dll
----------------------------------------
System.Core
    Assembly Version: 4.0.0.0
    Win32 Version: 4.6.1055.0 built by: NETFXREL2
    CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System.Core/v4.0_4.0.0.0__b77a5c561934e089/System.Core.dll
----------------------------------------
System.Xml
    Assembly Version: 4.0.0.0
    Win32 Version: 4.6.1064.2 built by: NETFXREL3STAGE
    CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System.Xml/v4.0_4.0.0.0__b77a5c561934e089/System.Xml.dll
----------------------------------------
Nevron.Licensing
    Assembly Version: 1.0.0.0
    Win32 Version: 16.5.16.12
    CodeBase: file:///C:/Program%20Files%20(x86)/Marlin3DprinterTool/Nevron.System.dll
----------------------------------------
Nevron.GraphicsGL
    Assembly Version: 16.5.16.12
    Win32 Version: 16.5.16.12
    CodeBase: file:///C:/Program%20Files%20(x86)/Marlin3DprinterTool/Nevron.GraphicsGL.DLL
----------------------------------------
Accessibility
    Assembly Version: 4.0.0.0
    Win32 Version: 4.0.30319.33440 built by: FX45W81RTMREL
    CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/Accessibility/v4.0_4.0.0.0__b03f5f7f11d50a3a/Accessibility.dll
----------------------------------------
System.Management
    Assembly Version: 4.0.0.0
    Win32 Version: 4.6.1055.0 built by: NETFXREL2
    CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System.Management/v4.0_4.0.0.0__b03f5f7f11d50a3a/System.Management.dll
----------------------------------------

************** JIT Debugging **************
To enable just-in-time (JIT) debugging, the .config file for this
application or computer (machine.config) must have the
jitDebugging value set in the system.windows.forms section.
The application must also be compiled with debugging
enabled.

For example:

<configuration>
    <system.windows.forms jitDebugging="true" />
</configuration>

When JIT debugging is enabled, any unhandled exception
will be sent to the JIT debugger registered on the computer
rather than be handled by this dialog box.

psavva, 2016-11-17:
[image]

cabbagecreek, 2016-11-28:
Fixed in the Marlin Editor

closed 2016-11-28 by cabbagecreek