cabbagecreek/Marlin3DprinterTool #57: Choosing a Folder with no marlin - no message is raised

psavva, 2016-11-17:
When choosing a folder, which is NOT Marlin, there is no indication that anything is wrong (except for the exception raised, which I mentioned earlier in another incident raised)

At minimum, there should be a message warning the user that the configuration file was not specified.

You probably want to clear the current configuration that is also loaded in this case.

cabbagecreek, 2016-11-18:
I will check that the directory is containing malin.ino and configuration.h and arduino.exe.
If not a message will be shown that its is not a directory for Marlin FW or ArduinoIDE

cabbagecreek, 2016-11-28:
I have fixed this Issue and testing that configuration.h and marlin.ino is existing

closed 2016-11-28 by cabbagecreek