cabbagecreek/Marlin3DprinterTool #59: Defining bed size

Pipshag, 2016-11-18:
I can't seem to figure out how to define my bed max positions for use when scanning the surface or probing.

My build plate is 300x200 (with software MAX position endstops) and only uses min-endstops for X, Y and Z. My probe has an offset in Y-depth. I have gone through and positioned the extruder and probe locations and saved them. When I switch over to probe the bed surface the software doesn't respect the locations but instead hit the wall on both X and Y. Lower right works since its 0,0.

It would be nice to be able to write in the positions in X,Y-format instead of actually moving the toolhead. This would also give a confirmation that the positions have been updated. As of now I can't be sure what's happened.

Pipshag, 2016-11-18:
It seems that the values are both displayed and saved without decimals.

Saving Lower Left corner to real position X10 Y0 boths displays and saves as X1000. See the communication log below when moving to Lower Left after saving it:

Send G1 X1000 Y0
Send M114
Receive ok

cabbagecreek, 2016-11-18:
I belive that you live in Sweden.. Phone me!  070 2627224

2016-11-18 9:32 GMT+01:00 Carl-Christian Ljunggren <notifications@github.com

Pipshag, 2016-11-18:
Changing Windows decimal formatting (Regional options) to "," instead of "." solves the problem.

cabbagecreek, 2016-11-18:
This may be a generic problem.. I will make a search for this potential problem on all paring of Marlin values

cabbagecreek, 2016-11-19:
Standard is something a developer like. Thats why each country has develop their own standard on decimal, date and timezone.

From now all decimals in M3DpT will use decimal-point = dot regardless of country

Pipshag, 2016-11-19:
That's great to hear. Suppose we can close the issue now?

cabbagecreek, 2016-11-19:
All changes done.. Tested Bed Level, Z-probe calculations, G29 with ABL/MBL
Looks like its working with both sweden,german,french and us 

This is a big change and I need some extra testing on this feauture!!!

Pipshag, 2016-11-19:
So, I have tested what I could thorougly and right off the bat an error occured. Since I switched my reigonal settings to use "." as a decimal point (but still keeping the swedish locale) I encountered the same problem.

When switching over to the default regional settings for Sweden, I could now configure the bed correctly.

Following has been tested working:
**Configuration Bed**
**Scan buildsurface**

What's not working is:
**Bed Level** (but this has never worked for me)
**Mash Bed Level** (but I recall this funciton not being active yet)

Haven't tested:
PID, since I use different PIDs for different extruders (supported now in Marlin and set with E0 or E1 but no information shown about this with M501).

I did uninstall the old version and reinstall this one.

cabbagecreek, 2016-11-20:
I have only done one extruder in the PID feature. I want that working flawless.. Then the differance is to add a GUI control for extruder 2.

The issue should havde been solved.. I will contact Pipshag and verify

Pipshag, 2016-11-20:
This is working correctly now, closing.

closed 2016-11-20 by Pipshag