cabbagecreek/Marlin3DprinterTool #60: Suggestion: Able to control Z-jump on probing

Pipshag, 2016-11-18:
I would want to be able to specifiy the Z-jump height when probing the bed. I do not use "Engage the Z-probe" since it's a fixed probe with enough clearance already.

The reasoning is that it's quite unnecessary to jump 10 mm on probe moves when I normally use 2 mm.

cabbagecreek, 2016-11-18:
This is for safety reason..  I have banged to many expensive sensors in the
bed and both distroyed sensors, mounting and nozzles.

I will look at this at make it changeable with a default of 10mm... When
the function is working it can be reduced to 2mm

Pipshag, 2016-11-18:
Thanks! The scanning takes too long time for rerunning it. Had to optimize it when I ran ABL 4x4.

cabbagecreek, 2016-11-20:
The amount of lift between probing is added to "Configure z-probe".. I start with 50 and change to 5 and the 1.. It has to be at least 1.

Pipshag, 2016-11-20:
Great, that's awesome!

cabbagecreek, 2016-11-20:
I belive that we can close this one.. 13 left to go ;-)

closed 2016-11-20 by cabbagecreek