cabbagecreek/Marlin3DprinterTool #61: Breaking: Bed level not making suggestions

Pipshag, 2016-11-18:
Since decimals are not working in Configuration Bed I added them to the XML-config instead.

`  <bed bedtype="4point" />
  <LowerLeftCorner x="10.00" y="0" />
  <LowerRightCorner x="264.00" y="0" />
  <UpperRightCorner x="264.00" y="148.00" />
  <UpperLeftCorner x="10.00" y="148.00" />
  <LowerLeftAdjuster x="12.00" y="0.00" />
  <LowerRightAdjuster x="262.00" y="0" />
  <UpperRightAdjuster x="262.00" y="148.00" />
  <UpperLeftAdjuster x="12.00" y="148.00" />
  <SafeHome x="135.00" y="68.00" />`

This fixes the positions and the software can now define the bed coordinates correctly. I then go over to **Bed Level** and press Probe the bed which initiates the probing. I then expect to see turns required to level the bed - but nothing changes. It just states "Unknown turn". I even did a "Auto Bed Level (G29)" which outputs the grid as expected but still no screw turns are displayed.

I have also gone in under **Scan buildsurface** and can probe and see the measured plane of the bed so everything here is working as it should.

Pipshag, 2016-11-18:
Might be related to #59 ?

Pipshag, 2016-11-18:
Probably not related, fixed #59 with changing locale settings. Bed Level still not giving screw turning suggestions (both Probe + G29). Same after running Scan buildsurface.

Pipshag, 2016-11-19:
Still a problem. Not sure if this is an active feature, documentation fuzzily notes "Bed Level Tool" as a feature.

I would also suggest adding pitch to the screws, 0.5 or 0.6 should make a difference, right?

Pipshag, 2016-11-19:
[image]

cabbagecreek, 2016-11-28:
This is fixed in new versions. The probe-offset should be within 35mm in old versions.
In the new version the probe-position is recalculated and should be within 3.14mm. (Normally this is within +-1mm and error is made by rounding double numbers to int. The error is multiplied with PI  ;-)

closed 2016-11-28 by cabbagecreek