cabbagecreek/Marlin3DprinterTool #77: Error Upon launching the new 0.12 Beta Version

psavva, 2016-12-09:
Application: Marlin3DprinterTool.exe
Framework Version: v4.0.30319
Description: The process was terminated due to an unhandled exception.
Exception Info: System.IO.FileNotFoundException
   at Marlin3DprinterTool.FrmMarlin3DprinterTool.InitializeComponent()
   at Marlin3DprinterTool.FrmMarlin3DprinterTool..ctor()
   at Marlin3DprinterTool.Program.Main()

Faulting application name: Marlin3DprinterTool.exe, version: 1.0.0.0, time stamp: 0x5847d9b2
Faulting module name: KERNELBASE.dll, version: 6.3.9600.18340, time stamp: 0x5736541b
Exception code: 0xe0434352
Fault offset: 0x00014878
Faulting process ID: 0x1db4
Faulting application start time: 0x01d2524364268a9c
Faulting application path: C:\Program Files (x86)\Marlin3DprinterTool\Marlin3DprinterTool.exe
Faulting module path: C:\Windows\SYSTEM32\KERNELBASE.dll
Report ID: a1f215ba-be36-11e6-8314-90004ec16484
Faulting package full name: 
Faulting package-relative application ID: 
…

cabbagecreek, 2016-12-09:
I belived that this is fixed.. I have sent new versions to the people that
got this problem.

The faulty MSI was first uploaded to Git, but some hours later a version
that fix this problem was uploaded.

Do a uninstall and be sure that c:\program x86\MarlinprinterTool director
is removed or empty
Reinstall from a fresh uploaded MSI (the new vith this problem fixed)

cabbagecreek, 2016-12-15:
+psavva 
Please redo the installation using v0.14-beta..
Close this issue if the problem is fixed

closed 2016-12-15 by psavva