cabbagecreek/Marlin3DprinterTool #79: Scrolling in the communications window is backwards.

cbitterfield, 2016-12-13:
(The first stays on top). It should work like tail.
Also I see a M119 constantly being sent. (You really can't read the logs. 
Probably need a pause on scroll.
[image]

cabbagecreek, 2016-12-13:
I am working on the Marlin Editor to make the migration to Marlin RC8 as easy as possible. The RS232 monitor uses part of the same code.  

By looking at the screenshoot I can tell that you are using an older version.

v0.12-beta have a new checkbox to remove the "busy processing" messages. (It can be removed in Firmware aswell)

There is also a commandline where you can send commands and the combobox stores all commands within a connection.

[image]

cabbagecreek, 2016-12-13:
The M119 is to check Endstop. For each responnce from M119 there is a new M119 sent. 
This is just the case when Endstop TAB is active. If some other tab is active the M119/Endstop is mute

cabbagecreek, 2016-12-15:
+cbitterfield 
Can this be closed?

closed 2017-01-10 by cabbagecreek