cabbagecreek/Marlin3DprinterTool #85: Display Software Version - Enhancement

cbitterfield, 2016-12-13:
Show software version (and build number) in Window header or about box

cabbagecreek, 2016-12-13:
This is available in v0.12-beta. Do you have the latest version installed?
[image]

There is also setters for some values.

If you yse Read EEPROM you get almost all parameters as you get when the RS232 connection to the printer is done.

cabbagecreek, 2016-12-15:
Both Marlin3DprinterTool version (In titlebar) and Marlin FW version is visible in v0.14-beta

[image]

+cbitterfield
Close this issue if the new version is working

closed 2017-01-09 by cabbagecreek

cbitterfield, 2017-01-09:
I loaded the Beta (Once I figured out that it was further along on the code base). Installer says it is not.

Still having some issues with it. Lots of long delays. When I have a chance to revisit, I’ll provide some better feedback

-------------------------------------------
Colin Bitterfield
mailto:colin@bitterfield.com
phone: (571) 533-4700
https://www.linkedin.com/in/colin-bitterfield-80612327
-------------------------------------------

cabbagecreek, 2017-01-09:
Have you downloaded the v0.16-beta?

cbitterfield, 2017-01-09:
No, but I will tonight.

-------------------------------------------
Colin Bitterfield
mailto:colin@bitterfield.com
phone: (571) 533-4700
https://www.linkedin.com/in/colin-bitterfield-80612327
-------------------------------------------

cabbagecreek, 2017-01-09:
Read the instructions on the release page.. This 0.16-beta has so many new
features yhat its almost a totaly new program

cbitterfield, 2017-01-10:
OK

-------------------------------------------
Colin Bitterfield
mailto:colin@bitterfield.com
phone: (571) 533-4700
https://www.linkedin.com/in/colin-bitterfield-80612327
-------------------------------------------