cabbagecreek/Marlin3DprinterTool #91: Bed leveling not showing all 4 locations (v15)

Pipshag, 2016-12-27:
When doing a bed level, the adjustment screw displays only 3 corners to adjust and not the 4th. After doing a manual adjustment (watching the G30 output and adjusting accordingly) the 4th one starts to show up and work as well. Is this related to a distance being to great? Doesn't seem to work all that reliably and always gives me CW-adjustment icons.

Tested on Marlin RC8.

cabbagecreek, 2016-12-27:
I will do a total code overview of the bed surface and surfacescan

cabbagecreek, 2016-12-28:
The G30 command is very different in rc7/8. I am working on a new component that do the G30 with G1/M119 commands ( = independent of FW). First step is Done on the "virtual gauge" used for z-Probe-offset. 

Skickat från min iPhone

closed 2017-02-01 by cabbagecreek