cabbagecreek/Marlin3DprinterTool #104: STL preview doesn't work for non-priviledged users

Anton-V-K, 2017-01-14:
I've installed 0.16 beta on Windows 7 64-bit SP1 hoping it will show STL thumbnails.
However all my attempts to make it work failed (well, not all - thumbnails are shown for admin only).
Initially I thought it could conflict with netfabb Basic, because the latter shows its icon in thumbnails, but after some experiments I tend to think the tool can't show thumbnails for non-privileged users.
I've read "C:\Program Files (x86)\Marlin3DprinterTool\STL thumbnail debug.rtf" but it doesn't explain what could block thumbnails from showing - the server is registered, and preview is shown on the tab (ServerManager.exe is ran with admin rights).

Could you please confirm whether thumbnails should be working for non-admin users as well?

cabbagecreek, 2017-01-16:
Thank you for a marvelous issue!!!!

ShellExtension is deep burried in Windows. There is alot of "gymnasics" insde windows to make this work (at all), I am not supriced that there still is some unvovered areas and that this powerful feature will have bugs. The later versions of Windows have more security than XP and Windows 7.

The STlhumbnail DLL is written so it should extend the Shell for Explorer and that is for all users. The problem may be that the installer have put admin priviledge on files in the ProgramFiles(86)\Marlin3DprinterTool.. Try to change the priviledge for all files so all users are able to read/write/execute.

There is a program called SystemManager.exe that have some debugging tools build in. Its tha same as do this from within M3DpT but more extended for debugging. Try to run that and open different testshell to see if the program gives more hints (or even crashes with some evidence of why..)

I need more info about Windows version to try to create a testenvironment with netfab installed.

Send me a mail on c a b b a g e c r e e k (a) g m a i l . c o m
and I will send you a license that hide the demo message from the Serial component that I use.

Many Regards

Anton-V-K, 2017-01-17:
Thanks for the reply. From it I can conclude that thumbnails should work for all users, so apparently something blocks this feature on my PC.
1. I doubt it is netfabb Basic who to blame. It has installed its icons only for one Windows user. Other non-privileged users see some standard Windows icon related to certificates (by default Windows treats STL-files as Certificate Trust List or something like this).
2. To be precise I use Windows Home Premium 64-bit SP1 (Russian).
3. I've changed NTFS-permissions to allow Users to modify files/folders under "C:\Program Files (x86)\Marlin3DprinterTool", but it has no positive effect.
4. I've enabled default Administrator account, and the Explorer shows default Windows icon for STL-files, i.e. no thumbnails => apparently your software registers "thumbnails server" only for the user, who starts the tool, and only privileged user can do it.
5. I've played a bit with Server Manager:
a) Open Test Shell doesn't show STL previews (should it?)
b) Test Server in Test Shell shows STL previews (works as expected)
c) Shell Debugger shows thumbnails (works as expected)
If any other information is needed to help you investigate the issue, let me know.

cabbagecreek, 2017-01-18:
The solution is to abandon the STL thumbnail Class.. It has to much problem

I have done a new STL viewer with the commersial component EZshellExtensions.
The cost for this is 99$ and I waiting for some donations to finance this.

With this component its possible to do some more with Windows Explorer that helps in "the-djungle-of-unorganized-STL"  ;-)

I have a pre v0-17beta for the people that are willing to test the Feature in Trial state = some AD will appear instead of the thumbnail/preview

Donators that use the PayPal button or join at Patreon (https://www.patreon.com/user?u=3753058) will get 2 hours help with Marlin Firmware thru mail and/or TeamViewer/Skype

cabbagecreek, 2017-01-23:
Test with v0.17-beta and see if we can close this issue

closed 2017-02-01 by cabbagecreek

Anton-V-K, 2017-02-08:
Thanks for taking care of this issue.
I've uninstalled 0.16 and installed 0.18 on the same system (Windows 7 64-bit SP1 Russian).
STL thumbnails works for the administator (the user who usually installs software), but non-priviledged users still have no luck. Actually with the previous version I had the same behavior... or the old settings still have the effect.
BTW the configuration tool `Marlin3DprinterToolStlConfiguration.exe` still requires "super-user" rights.

cabbagecreek, 2017-02-08:
I tried to isolate the installation for the STL viewer to bi within
Marlin3DprinterToolStlConfiguration.exe.
All the other programs dont need any priviledges..

Are you a programmer or just a user with knowledge.. I may need some help
with this

Anton-V-K, 2017-08-30:
Sorry for not replying earlier - had no time then, and then just forgot about it...

I tried the most recent available version (0.21-Beta) on the same OS... and unfortunately STL thumbnails don't work out of the box for non-privileged users.
Then I've started the configurator `Marlin3DprinterToolStlConfiguration` (it asked me for the administrator password - OK) and pressed the magic button `Install and Register STL Thumnail`. The tool showed me the message that everything went fine (`Registration done with Shell Register `)... and then a huge message box popped up (without any caption). It is hard to tell what it is all about, since the message got all non-latin characters scrambled (are you building with UTF support?) - probably something is wrong with the image.
Here is the message:
```
Running as : 64-bit

Processing file : C:\Program Files (x86)\Marlin3DprinterTool\Marlin3DprinterToolStlThumbnail.dll

COM Interop registeration of assembly failed!System.BadImageFormatException: ЌҐ г¤ «®бм § Јаг§Ёвм д ©« Ё«Ё бЎ®аЄг "file:///C:\Program Files (x86)\Marlin3DprinterTool\Marlin3DprinterToolStlThumbnail.dll" «ЁЎ® ®¤­г Ё§ Ёе § ўЁбЁ¬®бвҐ©. Ѓл«  б¤Ґ« ­  Ї®ЇлвЄ  § Јаг§Ёвм Їа®Ја ¬¬г, Ё¬Ґойго ­ҐўҐа­л© д®а¬ в.

€¬п д ©« : 'file:///C:\Program Files (x86)\Marlin3DprinterTool\Marlin3DprinterToolStlThumbnail.dll'

   ў System.Reflection.RuntimeAssembly._nLoad(AssemblyName fileName, String codeBase, Evidence assemblySecurity, RuntimeAssembly locationHint, StackCrawlMark& stackMark, IntPtr pPrivHostBinder, Boolean throwOnFileNotFound, Boolean forIntrospection, Boolean suppressSecurityChecks)

   ў System.Reflection.RuntimeAssembly.InternalLoadAssemblyName(AssemblyName assemblyRef, Evidence assemblySecurity, RuntimeAssembly reqAssembly, StackCrawlMark& stackMark, IntPtr pPrivHostBinder, Boolean throwOnFileNotFound, Boolean forIntrospection, Boolean suppressSecurityChecks)

   ў System.Reflection.RuntimeAssembly.InternalLoadFrom(String assemblyFile, Evidence securityEvidence, Byte[] hashValue, AssemblyHashAlgorithm hashAlgorithm, Boolean forIntrospection, Boolean suppressSecurityChecks, StackCrawlMark& stackMark)

   ў System.Reflection.Assembly.LoadFrom(String assemblyFile)

   ў RegisterExtension.RegAsmInvoker.RegasmAssembly(String assembly)

ЏаҐ¤гЇаҐ¦¤Ґ­ЁҐ: аҐЈЁбва жЁп ЇаЁўп§ЄЁ бЎ®а®Є ўлЄ«озҐ­ .

—в®Ўл ўЄ«озЁвм аҐЈЁбва жЁо ®иЁЎ®Є ЇаЁўп§ЄЁ бЎ®а®Є, гбв ­®ўЁвҐ §­ зҐ­ЁҐ Ї а ¬Ґва  аҐҐбва  [HKLM\Software\Microsoft\Fusion!EnableLog] (DWORD) ў 1.

ЏаЁ¬Ґз ­ЁҐ. ђҐЈЁбва жЁп ®иЁЎ®Є ЇаЁўп§ЄЁ бЎ®а®Є ¬®¦Ґв ЇаЁўҐбвЁ Є ­ҐЄ®в®а®¬г б­Ё¦Ґ­Ёо Їа®Ё§ў®¤ЁвҐ«м­®бвЁ.

—в®Ўл ®вЄ«озЁвм нвг дг­ЄжЁо, г¤ «ЁвҐ Ї а ¬Ґва аҐҐбва  [HKLM\Software\Microsoft\Fusion!EnableLog].
```
Here is the screenshot:
[image]

The scenario to chase this issue is as follows:
1. Install the application as Administrator.
2. Switch to an ordinary user (without administrator privileges) and try to bring thumbnails to life.

If you can't reproduce it this way, I'll test it on my other PCs.

I've also tried another button `STL Viewer prefered STL viewer` - I guess, it should register your STL Viewer (which works fine as a standalone application) as a default STL viewer... and the magic doesn't happen: STL are still being opened in other application (netfabb in my case).
So apparently the configurator applies (some of) the changes only to the user which had started it.

cabbagecreek, 2017-08-31:
You have to configute STL and register STL shell extension Thumbnails
Provider as Administrator. You cant switsh to a non-priviledged user and do
this.

The installation must be done bu a user in the Administrator group.

This becomes a problem on computers where you are not administrator.
Example where this is the case is company own computers where the company
dont want "unknown" software to be installed.

rgjones33, 2017-11-11:
Can the installer be an exe so we can run as Administrator?

cabbagecreek, 2017-11-14:
The msi should ask to be run as Administratpr.
This is needed to install the STL thumbnail

Excplain more why you suggest to change msi to exe. Are you a programmer
that want to help. I would like that.. The WIX installer is sometimes a bit
difficult

Many regards