cabbagecreek/Marlin3DprinterTool #116: paypal err

asigraph, 2017-02-10:
hi
i get err on donet buton &cenot make donetion

thenks

cabbagecreek, 2017-02-10:
I think the problem is oexploaterat.exe

I Will changera this to use prefered web browser.

cabbagecreek, 2017-02-11:
The update is done.. Will be available at 0.19-beta and I think I will release that version during the weekend

asigraph, 2017-02-14:
did u fix it?
i want to get the key but how to pay?

cabbagecreek, 2017-02-15:
I have not released the v0.19-beta yet..

Send me a mail with name and mail-adress and I will send you the license key..

Another way to do a donation is to use Patreon:
https://www.patreon.com/user?u=3753058

I will soon start 2 other project on Patreon:
* The NEMA17 bowden motor. A gear-reduction that has the dimension of a NEMA17 motor and a flexi-drive. This "dummy-Nema17" moror replaces the extruder NEMA17. The "real-NEMA17" is mounted on the frame and the flexi-shaft transfer the force. This NEMA17 solution is about 10 times stronger than the direct driven.

* The touchless filament sensor. Counts the filament used on spool and on the actual print.
The LCD display shows the counters. The sensor is connected to Marlin thru I2C and can get its commands from GCODE (=reset part-filament counter at start of print)
The sensor can meassure the width of the filament and report thar to Marlin as volt where 1.83mm = 1.83V. 
The sensor can reacht on "end-of-filament" and send the signal to Marlin.
The sensor can detect that filament is not moving and send the signal to Marlin.

Psyc0Flyer, 2017-02-15:
I have tried to donate also. What is your email?

cabbagecreek, 2017-02-15:
There is a bug if you have another prefered webbrowser and some kind of antivirus.  ( iexplorer have alot of mysterious jox. Do anyone use it anymore?)

On v0.19-beta the donation button use the prefered webbrowser. 

C a b b a g e C r e e k (a) g m a i l . C o m

Its possible to donate thru patrion. I think the pledge is even better there 😇

Skickat från min iPhone

asigraph, 2017-02-15:
Hi
My mail is  asigraph@gmail.com
Thenx

cabbagecreek, 2017-02-28:
The v0.19-beta should use the prefered webbrowser.

This should solve the problem.  I colese the issue when I get the first donation from Paypal

JackWetson, 2017-03-01:
Version 19 still contains this error.

`See the end of this message for details on invoking 
just-in-time (JIT) debugging instead of this dialog box.

************** Exception Text **************
System.IO.FileNotFoundException: Could not load file or assembly 'PayPal, Version=1.0.0.0, Culture=neutral, PublicKeyToken=null' or one of its dependencies. The system cannot find the file specified.
File name: 'PayPal, Version=1.0.0.0, Culture=neutral, PublicKeyToken=null'
   at Marlin3DprinterTool.FrmMarlin3DprinterTool.btnPayPal_Click(Object sender, EventArgs e)
   at System.Windows.Forms.Control.OnClick(EventArgs e)
   at System.Windows.Forms.Button.OnClick(EventArgs e)
   at System.Windows.Forms.Button.OnMouseUp(MouseEventArgs mevent)
   at System.Windows.Forms.Control.WmMouseUp(Message& m, MouseButtons button, Int32 clicks)
   at System.Windows.Forms.Control.WndProc(Message& m)
   at System.Windows.Forms.ButtonBase.WndProc(Message& m)
   at System.Windows.Forms.Button.WndProc(Message& m)
   at System.Windows.Forms.NativeWindow.Callback(IntPtr hWnd, Int32 msg, IntPtr wparam, IntPtr lparam)

WRN: Assembly binding logging is turned OFF.
To enable assembly bind failure logging, set the registry value [HKLM\Software\Microsoft\Fusion!EnableLog] (DWORD) to 1.
Note: There is some performance penalty associated with assembly bind failure logging.
To turn this feature off, remove the registry value [HKLM\Software\Microsoft\Fusion!EnableLog].

************** Loaded Assemblies **************
mscorlib
    Assembly Version: 4.0.0.0
    Win32 Version: 4.6.1586.0 built by: NETFXREL2
    CodeBase: file:///C:/Windows/Microsoft.NET/Framework64/v4.0.30319/mscorlib.dll
----------------------------------------
Marlin3DprinterTool
    Assembly Version: 0.18.6248.14495
    Win32 Version: 0.18.0.0
    CodeBase: file:///C:/Program%20Files%20(x86)/Marlin3DprinterTool/Marlin3DprinterTool.exe
----------------------------------------
System.Windows.Forms
    Assembly Version: 4.0.0.0
    Win32 Version: 4.6.1586.0 built by: NETFXREL2
    CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System.Windows.Forms/v4.0_4.0.0.0__b77a5c561934e089/System.Windows.Forms.dll
----------------------------------------
System
    Assembly Version: 4.0.0.0
    Win32 Version: 4.6.1586.0 built by: NETFXREL2
    CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System/v4.0_4.0.0.0__b77a5c561934e089/System.dll
----------------------------------------
System.Drawing
    Assembly Version: 4.0.0.0
    Win32 Version: 4.6.1586.0 built by: NETFXREL2
    CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System.Drawing/v4.0_4.0.0.0__b03f5f7f11d50a3a/System.Drawing.dll
----------------------------------------
Marlin3DprinterToolConfiguration
    Assembly Version: 1.0.0.0
    Win32 Version: 1.0.0.0
    CodeBase: file:///C:/Program%20Files%20(x86)/Marlin3DprinterTool/Marlin3DprinterToolConfiguration.DLL
----------------------------------------
MarlinComunicationHelper
    Assembly Version: 1.0.0.0
    Win32 Version: 1.0.0.0
    CodeBase: file:///C:/Program%20Files%20(x86)/Marlin3DprinterTool/MarlinComunicationHelper.DLL
----------------------------------------
ZylSerialPort
    Assembly Version: 1.66.6164.31970
    Win32 Version: 1.66.6164.31970
    CodeBase: file:///C:/Program%20Files%20(x86)/Marlin3DprinterTool/ZylSerialPort.DLL
----------------------------------------
System.Windows.Forms.DataVisualization
    Assembly Version: 4.0.0.0
    Win32 Version: 4.6.1586.0
    CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System.Windows.Forms.DataVisualization/v4.0_4.0.0.0__31bf3856ad364e35/System.Windows.Forms.DataVisualization.dll
----------------------------------------
LedBulbUserControl
[... 66 lines cut ...]
    Win32 Version: 4.6.1586.0 built by: NETFXREL2
    CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/Accessibility/v4.0_4.0.0.0__b03f5f7f11d50a3a/Accessibility.dll
----------------------------------------
System.Data
    Assembly Version: 4.0.0.0
    Win32 Version: 4.6.1636.0 built by: NETFXREL3STAGE
    CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_64/System.Data/v4.0_4.0.0.0__b77a5c561934e089/System.Data.dll
----------------------------------------

************** JIT Debugging **************
To enable just-in-time (JIT) debugging, the .config file for this
application or computer (machine.config) must have the
jitDebugging value set in the system.windows.forms section.
The application must also be compiled with debugging
enabled.

For example:

<configuration>
    <system.windows.forms jitDebugging="true" />
</configuration>

When JIT debugging is enabled, any unhandled exception
will be sent to the JIT debugger registered on the computer
rather than be handled by this dialog box.`

cabbagecreek, 2017-03-01:
I will do a hotfix on this.

JackWetson: Send me am email at c a b b a g e c r e e k (a) g m a i l . c o m

an d I will fix this ASAP

cabbagecreek, 2017-03-01:
JackWetson

I have tested this and it worked with Chrome..

Look in the C:\program (x86)\marlin3dprintertool directory and se if you see the file PayPal.dll.

My suggestion is to uninstall marlin3Dprinter and delete all files inside  C:\program (x86)\marlin3dprintertool directory

Then reinstall the MSI again..

contact me thru mail , facebook or G+/hangout. We fix this in notime ;-)

JackWetson, 2017-03-01:
OK, so I did an un-install which removed the folder fully anyway. Paypal.ddl now exists however a new error

`System.InvalidOperationException: Cannot start process because a file name has not been provided.
   at System.Diagnostics.Process.Start()
   at System.Diagnostics.Process.Start(ProcessStartInfo startInfo)
   at System.Windows.Forms.Control.OnClick(EventArgs e)
   at System.Windows.Forms.Button.OnClick(EventArgs e)
   at System.Windows.Forms.Button.OnMouseUp(MouseEventArgs mevent)
   at System.Windows.Forms.Control.WmMouseUp(Message& m, MouseButtons button, Int32 clicks)
   at System.Windows.Forms.Control.WndProc(Message& m)
   at System.Windows.Forms.ButtonBase.WndProc(Message& m)
   at System.Windows.Forms.Button.WndProc(Message& m)
   at System.Windows.Forms.NativeWindow.Callback(IntPtr hWnd, Int32 msg, IntPtr wparam, IntPtr lparam)`

cabbagecreek, 2017-03-01:
Ok.. 
I will create my own webbrowser instead of using the one that should be installed (iexplerer.exe) or any other prefered webbrowser.

This will be a very small webbrowser that only covers the need for PayPal..

I will have this don very soon

cabbagecreek, 2017-03-01:
I have done a Chrome webbrowser just for PayPal donations.. That part of M3DpT is 5 times bigger than the actual SW.

The new MSI us uploaded to Release without changing version.
You will have to unnstall and remove C:\program (x86)\Marlin3DprinterTool and then do a Installation with the new MSI.

cabbagecreek, 2017-03-14:
Skickat från min iPhone

cabbagecreek, 2017-04-06:
In ver 0.21-beta I think all PayPal problems are solved

closed 2017-04-06 by cabbagecreek