cabbagecreek/Marlin3DprinterTool #119: Probe Crash

Psyc0Flyer, 2017-02-16:
When using the Dock Z probe, it homes, goes to destination, probes, raises, reprobes and goes in for final probe BUT the probe plunger does not release to activate the probe and nozzle crashes into bed :/

cabbagecreek, 2017-02-17:
Is this a BLTouch probe?

Psyc0Flyer, 2017-02-17:
Yes.

-- 
Best regards!

Prometheus-3D
610-780-2714

This electronic transmission contains information which may be confidential
and/or privileged. This information is intended for use only by the
individual or entity named above and may not be shared elsewhere or
otherwise. If you are not the intended recipient (or the employee or agent
responsible for delivering this information to the intended recipient), you
are hereby notified that any use, dissemination, distribution, or copying
of this communication is prohibited. If you have received this information
in error, please notify me by electronic mail and delete all copies of the
transmission.

Thank you.

cabbagecreek, 2017-02-18:
Okey. I Will build a 3D printer that covers this setup today. I even might do a video of the setup and testing.

cabbagecreek, 2017-02-28:
The BLTouch version is released.. v0.19-beta This version covers the problem

closed 2017-04-06 by cabbagecreek