cabbagecreek/Marlin3DprinterTool #124: No Comp Port Found

alicuore, 2017-03-14:
Hi All,

I'm Ali from Malaysia. I've downloaded the latest version of Marlin3DprinterTool on my Windows using Parallels Desktop via my Mac.

The problem is, I couldn't connect my printer as there's no comp port found. Is there any way to make it happen? Or do I have to use a proper windows computer? Do I need additional driver?

Thank you in advance.

cabbagecreek, 2017-03-14:
I have used a component from ZylSoft that have the speed needed for communication with the 3D printer.
The 3D printer is not actually build to communicate back to the host. I use the responce back to the host and do interpetations of that to guide the printer.
The Markin firmware has no flow-control and the communication needs to be very fast.

This is one of the main reasons why I have choosen C#/.Net. I have tried alot of different platforms in the beginning of this project to find a platform that could be used for Mac, Linux and Windows. All my testing come to the conclusion that only C# had a communication that was fas enought.

If I do a multi-platform try again it will be trying with Xamarin...  That approach needs to totaly separate the GUI part from the program logic and do the GUI for each platform. This is an enormus task.

alicuore, 2017-03-17:
Hi Sir,

Thank you for your reply. I see, to be honest I don't really have a knowledge in this C#/.Net.....

I checked out the ZylSoft website, but sadly have to pay 100++USD for it. 

So what's the solution if I want to get the connection/communication from 3D printer and the software? If I download the Microsoft .Net on a proper windows computer, will it works?

Hope to hear soon from you. Thank you in advance

cabbagecreek, 2017-04-06:
If you use any Windows PC from Windows 7 to Windows 10 the program will work:
Look at Marlin3DprinterTool.se for more info

closed 2017-04-06 by cabbagecreek