cabbagecreek/Marlin3DprinterTool #128: 4/4 version crashes when run on W10 X64 machine.

CapnBob, 2017-04-05:
Uninstalled previous version; reinstalled ,same result.  

Problem signature:
  Problem Event Name:	CLR20r3
  Problem Signature 01:	Marlin3DprinterTool.exe
  Problem Signature 02:	0.20.0.0
  Problem Signature 03:	58e3a6bc
  Problem Signature 04:	Marlin3DprinterTool
  Problem Signature 05:	0.20.0.0
  Problem Signature 06:	58e3a6bc
  Problem Signature 07:	be
  Problem Signature 08:	10
  Problem Signature 09:	System.BadImageFormatException
  OS Version:	6.1.7601.2.1.0.256.48
  Locale ID:	1033
  Additional Information 1:	e6e6
  Additional Information 2:	e6e6c208b90db2aa028f8f653cabbf23
  Additional Information 3:	abe2
  Additional Information 4:	abe2ee164908dd9f34091cf63a7be37f

Read our privacy statement online:
  http://go.microsoft.com/fwlink/?linkid=104288&clcid=0x0409

If the online privacy statement is not available, please read our privacy statement offline:
  C:\Windows\system32\en-US\erofflps.txt

cabbagecreek, 2017-04-05:
I will look at this in a couple of hours. Possible to do thru teamviewer?

Skickat från min iPhone

CapnBob, 2017-04-05:
Sure,
      
      All set up for a one-timer.
      
      Send me a quick email when you're done.   
      
      
      
      
      On 04/05/2017 8:58 AM, Johnny Lindén wrote:
    
    I will look at this in a couple of hours. Possible to
      do thru teamviewer?
      
      Skickat från min iPhone

      http://go.microsoft.com/fwlink/?linkid=104288&clcid=0x0409
      thread.

cabbagecreek, 2017-04-05:
> 143418/17495839/a5054eac-5d88-11e6-95fc-7290892c7bb5.png","
> avatar_image_url":"https://cloud.githubusercontent.com/
> assets/143418/15842166/7c72db34-2c0b-11e6-9aed-b52498112777.png
> ","action":{"name":"Open in GitHub","url":"https://github.
> com/cabbagecreek/Marlin3DprinterTool"}},"updates":{"snippets":[{"icon":
> 128#issuecomment-291853068"}}}

Is it possible that you can test this MSI instead.. It looks like the MSI
have a missmatch between x86 and x64 code..

​
 Marlin3DprinterTool.msi
<https://drive.google.com/file/d/0B7H1HDErektQOGZPTTZOb3dNV2M/view?usp=drive_web>
​
In this version there is an automatic Download of upgrade MSI.. Answer NO
because the MSI that you will download is the same as you already have..
Its maked with version 0.22-beta but its the same as you get now = 0.21-beta

I this cure you problem I will do a release of this during the evening..

Regards

CapnBob, 2017-04-05:
Have completed it:
      
      Installed it; it asked if I wanted to upgrade to 0.22.0.0
      Did that, it downloaded to desktop ran that and got the same
      thing.
      Opened the exe and it says 21.0 is  installed?  
      
      That one is working, but it's,not 22...
      
      
      
      
      On 04/05/2017 10:29 AM, Johnny Lindén wrote:
    
    Is it possible that you can test this MSI instead.. It
      looks like the MSI
      have a missmatch between x86 and x64 code..
      
      ​
      Marlin3DprinterTool.msi
<https://drive.google.com/file/d/0B7H1HDErektQOGZPTTZOb3dNV2M/view?usp=drive_web>
      ​
      In this version there is an automatic Download of upgrade MSI..
      Answer NO
      because the MSI that you will download is the same as you already
      have..
      Its maked with version 0.22-beta but its the same as you get now =
      0.21-beta
      
      I this cure you problem I will do a release of this during the
      evening..
      
      
      Regards

      

      http://go.microsoft.com/fwlink/?linkid=104288&clcid=0x0409
      http://go.microsoft.com/fwlink/?linkid=104288\u0026clcid=0x0409\n\u003e

cabbagecreek, 2017-04-05:
> view?usp=drive_web>
> drive.google.com/file/d/0B7H1HDErektQOGZPTTZOb3dNV2M/
> view?usp=drive_web\u003e\n
> ​\nIn this version there is an automatic Download of upgrade MSI.. Answer
> fwlink/?linkid=104288\u0026clcid=0x0409\n\u003e \u003e\n\u003e \u003e If
> githubusercontent.com/assets/\n\u003e 143418/17495839/a5054eac-5d88-
> cloud.githubusercontent.com/\n\u003e assets/143418/15842166/
> fwlink/?linkid=104288\\u0026clcid=0x0409\\n\\u003e\n\u003e \\n\\u003e If
> Marlin3DprinterTool/issues/\n\u003e 128#issuecomment-291853068\"}}}\n\u003e\n\u003e
> cabbagecreek/Marlin3DprinterTool/issues/128#issuecomment-291870021\
> WvPD3bpks5rs54ugaJpZM4Mz8fk\u003e\n\u003e
> com/cabbagecreek/Marlin3DprinterTool/issues/128#issuecomment-291879307"}}}

You got a testversion.. The MSI that you update to is the same as the
v0.21-beta.. The uploaded version is just a fake 0.22-beta that I use for
testing

cabbagecreek, 2017-04-05:
> Do you have a working version installed.. keep answer NO to upgrades..
>
> I will upload the v0.21-beta release soon and correct the version on the
> upgrade
>
>
> 2017-04-05 17:24 GMT+02:00 Johnny Lindén <cabbagecreek@gmail.com>:

I think I will add a "Check for upgrade" in the setup..

2017-04-05 17:26 GMT+02:00 Johnny Lindén <cabbagecreek@gmail.com>:

CapnBob, 2017-04-05:
Yes, present version is working.
      
      
      
      On 04/05/2017 11:26 AM, Johnny Lindén wrote:
    
    Do you have a working version installed.. keep answer
      NO to upgrades..
      
      I will upload the v0.21-beta release soon and correct the version
      on the
      upgrade
      
      
      2017-04-05 17:24 GMT+02:00 Johnny Lindén
      <cabbagecreek@gmail.com>:
      
      > You got a testversion.. The MSI that you update to is the
      same as the
      > v0.21-beta.. The uploaded version is just a fake 0.22-beta
      that I use for
      > testing
      >
      >

      <https://drive.google.com/file/d/0B7H1HDErektQOGZPTTZOb3dNV2
      http://go.microsoft.com/fwlink/?linkid=104288&clcid=0x0409
      http://go.microsoft.com/fwlink/?linkid=104288\u0026clcid=0x0409\n\u003e
      https://github.com/cabbagecreek/Marlin3DprinterTool/issues/
      <https://github.com/cabbagecreek/Marlin3DprinterTool/issues/
<http://drive.google.com/file/d/0B7H1HDErektQOGZPTTZOb3dNV2M/view?usp=drive_web%5Cu003e%5Cn>
      http://go.microsoft.com/fwlink/?linkid=104288\u0026clcid=0x0409\n\u003e
      http://go.microsoft.com/fwlink/?linkid=104288\\u0026clcid=
<http://github.com/cabbagecreek/Marlin3DprinterTool/issues/128#issuecomment-291870021%5Cu003e,%5Cn%5Cu003e>

cabbagecreek, 2017-04-05:
Marvelous!!! 

I will try to make a new 0.21-beta release with this fix and reset the "LatestVersion" to v0.21-beta

(During test it was set to v0.22-beta to force update)

closed 2017-04-06 by cabbagecreek