cabbagecreek/Marlin3DprinterTool #137: No action when Auto Update is started

elfhelmp, 2017-04-17:
Hello,
I have installed licensed version 0.22.6315.19872 according to Auto Update and when a pop-up shows new release version 0.23.0 and hit Yes, nothing happens.

I also tried to manually download 0.23.0 but I can't find it anywhere, latest on github is still 0.22.0

Thanks,
elfhelm

cabbagecreek, 2017-04-17:
I Have set the version to 0.23 Just to test the Update function.. I am
currently working on this feature and I belive that I will be finished in a
hour or two

The Msi downloaded is the v0.22 so there is no Update.. but the update
function is working ;-)

I belive it will be faster than look at GitHub each time

Regards

cabbagecreek, 2018-04-21:
I have released v0.26-beta and this issue is solved

closed 2018-04-21 by cabbagecreek