cabbagecreek/Marlin3DprinterTool #138: Scan buildsurface

honniesr, 2017-04-25:
I am using a Wanhao i3 duplicator V2.1 with a bltouch mounted 42mm in front of the extruder nozzle with 0 X axis offset.

Marlin3DprinterTool Version is 0.22.6316.28050  

When I run Scan buildsurface, it starts at X0 Y0 and seems to be ignoring the zprobe offset and the probe_bed_position settings in the marlin config listed below.  I also set the probe offset to nozzle in the Z Probe Offset tab.

  #define Y_PROBE_OFFSET_FROM_EXTRUDER -42
  #define LEFT_PROBE_BED_POSITION 10
  #define RIGHT_PROBE_BED_POSITION 170
  #define FRONT_PROBE_BED_POSITION 10
  #define BACK_PROBE_BED_POSITION 155

Should I be doing something different?

Thanks,
Robert

cabbagecreek, 2017-04-27:
Marlin3DprinterTool and the scansurface is not depending on Marlin Firmware at all.
You have to configure the bed to set the points where the corners of the bed is "probable"

The scan then divides the distance betwwen the corners to create a mesh-point-list. 
It homes the printer using G28
Then it moves to the first point using G1 X Y
The probing is done by G30 and the result is stored in the mesh-point-list
When all pints are probed the values are adjusted to get maximum resolution on the surface-chart.
The surface-chart gets all the mesh-points and shows a visual "image" of how the probe has scanned the surface

honniesr, 2017-05-01:
How do you "configure the bed" to set the points where the corners of the bed are "probable"?

cabbagecreek, 2017-05-01:
THIS IS FOR MARLIN3DPRITERTOOL ONLY!!! 
All G! positions are based on wher the nozzle is. 
If the probe is 30mm to the left of the nozzle and the position of the nozzle is at 0,0 the probe will be 30mm outside the bed = Not "Probable". Moving the nozzle to 30,0 will get the probe at position 0,0 and thet position is "probable". The "Pin" button get the actual nozzleposition. The Save button saves that position. The "GoTo" button goes to the stored positions (= use as confirmation that the positions are "probeable") 

Short: Move the probe to points where the probe is able to read the bed and store the nozzle position.

cabbagecreek, 2017-09-26:
This issue is solved

closed 2017-09-26 by cabbagecreek