cabbagecreek/Marlin3DprinterTool #140: which stl viewer is required for the thumbnail generation?

DjDemonD, 2017-06-18:
Can you specify which stl viewer is needed for the thumbnail generationg (in Windows).

Many thanks
very useful feature.

cabbagecreek, 2017-06-18:
The reason why Windows File Explorer is not showing a thumbnail is because ther is no STL WindowsFileExplorerShellExtensionStlProvider..
So I created one. ;-)

The task is to hook in the StlProvider inside Windows.. That is something that is not easy to do. Windows is not willing to do that. The thumbnails are small "pictures" of the filecontent. To let windows file explorer accept a new StlProvider for those images you will have to delete all existing thumbnails.  Thats why it takes a while for windows File Explorer to recreate ALL thumbnails. Easy thumbnails like .TXT .DOC etc is a static picture but STL need to render a miniature image and assign that as the STL thumbnail. Thats why it takes so long time to create thumbnails for STL.

So back to the question.. How do I get the WindowsFileExplorerShellExtensionStlProvider..
Download the latest version here: 
www.Marlin3DprinterTool.se/LatestVersion/Marlin3DprinterTool.msi

Install the msi

In the start meny you will get a Marlin3DprinterTool folder
Inside that there is a Marlin3DprinterToolSTLconfiguration

Install the STL provider and do all trick with Windows File Explorer to "force" it to accept the nes STL thumbnail provider. 

Go to your STL folder. Notice that the titlebar has a progressbar.. Thats WindowsFileExplorer that creates new thumbnails.. This can take along time.. (if the folder conteains many large STL)

In the config there is a lightweight STL viewer thats able to show and turn a STL. If you assossiate the STL viewer for STL files this will become your prefered viewer. Athers have Simplify3D as prefered viewer but this is faster and you will have to right-click and "open with Simplify3d" to have the STL uploaded to simplify3D for printing. (Replace Simplify3D with Cura, or whatever 3Dprinting software you are using)

I should create new videos but I need a narrator that can get voice to the videos. I create a story-board text and a quick un-cut version of the video with LARGE empty spaces . The narrator talks thru the story-board and try to keep the pace of the video.
In the cut the voice and video will be synced and all extra  empty spece will be cut away to create a good flow in the video.

DjDemonD, 2017-06-18:
Thanks that's helpful.
So here is what I have done so far. I went to disc cleanup and deleted all my thumbnails. I associated your viewer with stls and when I double click an stl your viewer opens, but I am still only getting a fixed thumbnail.

cabbagecreek, 2017-06-19:
That was the old procedure.. This should be done inside the ConfigTool.

Is it possible to take the troubleshoot on mail and report the error/solution here?

c a b b a g e c r e e k @ g m a i l . c o m

I try to do a simple doc with screenshoots and guided walk-thru-installation

[STLthumbnail.docx](https://github.com/cabbagecreek/Marlin3DprinterTool/files/1084205/STLthumbnail.docx)

cabbagecreek, 2017-09-26:
This issue is solved..

closed 2017-09-26 by cabbagecreek