cabbagecreek/Marlin3DprinterTool #142: #error "LCD_BED_LEVELING requires MESH_BED_LEVELING or PROBE_MANUALLY."

EnricoSx, 2017-09-03:
nel file configuration_overall ho scritto
#define PROBE_MANUALLY
#define LCD_BED_LEVELING

ma mi da l'errore del titolo.
ps. c'è qualche documentazione su come si fa il livellamento con PROBE_MANUALLY e LCD?

cabbagecreek, 2017-09-04:
translation using Goole:

> 
in the file configuration_overall I wrote
#define PROBE_MANUALLY
#define LCD_BED_LEVELING

but I got the title error.
ps. is there any documentation on how to level up with PROBE_MANUALLY and LCD?

closed 2017-09-04 by cabbagecreek

cabbagecreek, 2017-09-04:
This is not a M3DpT issue.. Its a Marlin FW issue.

Documentation about Marlin FW can be found at: http://marlinfw.org/docs/configuration/configuration.html

Documentation about PROBE_MANUALLY defines that no probe is used. The M3DpT will use the probe in many of its features.

Some of the M3DpT features are useful, but every feature that helps configure bed-level is not working if you dont have a Z-probe.