cabbagecreek/Marlin3DprinterTool #147: Constant explorer.exe crashes when complex STL models are being rendered

RonGeorge, 2017-11-16:
I have the correct 64 bit installed, but I get crashing every 5 minutes or so when accessing a folder with STLs.

`Application: Explorer.EXE
Framework Version: v4.0.30319
Description: The process was terminated due to an unhandled exception.
Exception Info: System.IO.DirectoryNotFoundException
   at System.IO.__Error.WinIOError(Int32, System.String)
   at System.IO.FileStream.Init(System.String, System.IO.FileMode, System.IO.FileAccess, Int32, Boolean, System.IO.FileShare, Int32, System.IO.FileOptions, SECURITY_ATTRIBUTES, System.String, Boolean, Boolean, Boolean)
   at System.IO.FileStream..ctor(System.String, System.IO.FileMode, System.IO.FileAccess, System.IO.FileShare)
   at HelixToolkit.Wpf.ModelReader.Read(System.String)
   at Marlin3DprinterToolStlThumbnailProvider.StlThumbnailProvider+<>c__DisplayClass1_0.<GetThumbnailImage>b__0()
   at System.Threading.ExecutionContext.RunInternal(System.Threading.ExecutionContext, System.Threading.ContextCallback, System.Object, Boolean)
   at System.Threading.ExecutionContext.Run(System.Threading.ExecutionContext, System.Threading.ContextCallback, System.Object, Boolean)
   at System.Threading.ExecutionContext.Run(System.Threading.ExecutionContext, System.Threading.ContextCallback, System.Object)
   at System.Threading.ThreadHelper.ThreadStart()`

cabbagecreek, 2017-11-18:
I will try to fix this.. This happends when the STL is to complex. Fo you
have a STL file that crashes that I can test with?

Regards

RonGeorge, 2017-11-20:
Sorry, it took me a number of hours of trial and error to find some exact files that crash it every time. This zip of STLs will do it. https://1drv.ms/u/s!AkaMSvNd1E8DsgNpAph59wBPJpgU

Genae, 2018-01-09:
Hi, I have a similar problem. I have a folder filled with about 800 stl files with an average size of 2MB (the biggest one is 42MB). When I slowly scroll down the list to create 3 thumbnails at a time i get a crash after 10-40 thumbnails created. When I try it again most of the time the thumbnail will be created fine afterwards so it doesnt seem to be a problem with a specific file. Every crash gives me 2 errors:
`
Application: explorer.exe
Framework version: v4.0.30319
Description: The process was terminated due to an unhandled exception.
Exceptions: System.AccessViolationException
   at System.Windows.Media.Composition.DUCE + UnsafeNativeMethods.WgxConnection_SameThreadPresent (IntPtr)
   System.Windows.Media.Renderer.Render (IntPtr, Channel, System.Windows.Media.Visual, Int32, Int32, Double, Double, System.Windows.Media.Matrix, System.Windows.Rect)
   System.Windows.Media.Imaging.BitmapVisualManager.Render (System.Windows.Media.Visual, System.Windows.Media.Matrix, System.Windows.Rect)
   at HelixToolkit.Wpf.Viewport3DHelper.RenderBitmap (System.Windows.Controls.Viewport3D, System.Windows.Media.Brush, Int32)
   at HelixToolkit.Wpf.BitmapExporter.Export (System.Windows.Controls.Viewport3D, System.IO.Stream)
   at Marlin3DprinterToolStlThumbnailProvider.StlThumbnailProvider + <> c__DisplayClass1_0. <GetThumbnailImage> b__0 ()
   at System.Threading.ExecutionContext.RunInternal (System.Threading.ExecutionContext, System.Threading.ContextCallback, System.Object, Boolean)
   at System.Threading.ExecutionContext.Run (System.Threading.ExecutionContext, System.Threading.ContextCallback, System.Object, Boolean)
   at System.Threading.ExecutionContext.Run (System.Threading.ExecutionContext, System.Threading.ContextCallback, System.Object)
   at System.Threading.ThreadHelper.ThreadStart ()`

and:
`Name of the failed application: explorer.exe, version: 10.0.16299.192, timestamp: 0x494a4408
Name of the faulty module: RGB9Rast.dll, Version: 10.0.16299.15, Timestamp: 0xb50dbd94
Exception code: 0xc0000005
Error offset: 0x000000000001477b
ID of the failed process: 0x300c
Start time of the failed application: 0x01d389985536e86e
Path of the failed application: C: \ WINDOWS \ explorer.exe
Path of the failed module: C: \ WINDOWS \ SYSTEM32 \ RGB9Rast.dll
Report ID: 90eb0dde-bbdc-42f0-8fcd-70871b59b14b
Full name of the failed package:
Application ID that is relative to the failed package:`

I am running Windows 10 (64bit).
STL Files that cause the problem have been downloaded from: https://www.shapeways.com/designer/mz4250/creations

cabbagecreek, 2018-01-11:
I will have an extra look at this.. Is it possible that you can send me one
of the STL that craches the STL thumbnails?

Regards

cabbagecreek, 2018-01-15:
I am actually looking at this problem right now.
Send me STL files that are large and/or complex

2018-01-11 17:58 GMT+01:00 Johnny Lindén <cabbagecreek@gmail.com>:

cabbagecreek, 2018-01-15:
[Marlin3DprinterToolStlThumbnail.zip](https://github.com/cabbagecreek/Marlin3DprinterTool/files/1631967/Marlin3DprinterToolStlThumbnail.zip)
I have a solution!
You need to unregister the STL thumbnail provider using Marlin3DprinterToolStlConfiguration.
Reboot the PC.
Now the STL thumbnail provider is able to be replaced,
Unzip the attached DLL.
Replace the dll in C:\program x86\Marlin3DprinterTool directory
Register the DLL again using  Marlin3DprinterToolStlConfiguration

Now the crashing FileExlorer should be fixed..

If you get problems contact me thru mail/message and I will try to help

cabbagecreek, 2018-01-15:
I need complex STL files that crashed File Explorer with previous version

RonGeorge, 2018-01-15:
This link is still valid, full of STLs that crash. https://1drv.ms/u/s!AkaMSvNd1E8DsgNpAph59wBPJpgU

cabbagecreek, 2018-01-15:
Have you tried to replace the DLL two comments up?
Follow the instruction.
You must unistall the STL thumbnail Provider and reboot to be able to overwrite with the new DLL.
The reinstall the STL thumbnail Provider and Windows File Explorer should nor crash anymore.
RonGeorge please confirm if this new DLL is working

cabbagecreek, 2018-01-15:
I think all files are working..  RonGeorge: Please confirm that its working at your site aswell

[image]

RonGeorge, 2018-01-15:
I deregistered, rebooted, replaced the dll and then went back into marlinconfig and tried to register and got this.
[image]

cabbagecreek, 2018-01-15:
The line:

> ...FileLoadException; Could not load file or assembly.. 
Indicates that the DLL is not found..

I post a total MSI file instead. 
Uninstall M3DpT and remove the C:\program x86\Marlin3DprinterTool directory.

Install the new MSI file.. 
Rerun the Marlin3DprinterSTLconfiguration

[Marlin3DprinterTool.zip](https://github.com/cabbagecreek/Marlin3DprinterTool/files/1632878/Marlin3DprinterTool.zip)

cabbagecreek, 2018-01-15:
If this new MSI is working for RonGeorg i will publish it so everyone get this new 0.24-beta version.

After that RonGeorg has confirmed that this is working it will be available at http://www.Marlin3DprinterTool.se/LatestVersion/Marlin3DprinterTool.msi

The build-in AutoUpdater should see that a newer version is available and try to download and install that version. If this is working there is no need to do the manual download and install.

RonGeorge, 2018-01-15:
It seems like it is working. I installed with the msi and then set the color to purple. I went to the folder that I gave you and it rendered thumbnails. The only thing, and I don't care about this, but it might be an indication of something down the road. It 'lost' the color choice when I went to the next folder and it reverted to white. I sat at this new folder for about 9 minutes or so and it didn't crash, but it did seem "stuck". So I just closed the window, re-opened it and it started rendering again in white. So, I think it works.

cabbagecreek, 2018-01-15:
MARVELOUS!!!!
Have you ever used white as render color.. The default is blue so I have no idea where the white came from. IMHO white is not a good choice for color because Windows FileExplorer use sometimes white as background.

The thing about Thumbnail Shell Extension is that its deep burried inside Windows and Windows File Explorer is just an application on top of Windows. Thats the reason why the rendered thumbnails are available in other program aswell (like Simplify3d)

When this thumbnail provider has done some rendering its like "stuck" in the hidden dungeons inside windows. Sometimes it finds cached thumbnails from previous sessions and start use them. Its almost unpredictable how this is done.

BUT.. when its working, it keeps on working and becomes better and faster over time ;-)

I hope that everyone enjoys this bugfix 

I will be posting this as the new 0.24-beta version for the auto-updater to pickup

RonGeorge, 2018-01-15:
Ok, I spoke too soon.. it just crashed BUT it crashed after it rendered the thumbnails. I do think this is a vast improvement though and excellent work. Here are the event viewer logs.
```

Error .NET Runtime
Application: explorer.exe
Framework Version: v4.0.30319
Description: The process was terminated due to an unhandled exception.
Exception Info: System.AccessViolationException
   at System.Windows.Media.Composition.DUCE+UnsafeNativeMethods.WgxConnection_SameThreadPresent(IntPtr)
   at System.Windows.Media.Renderer.Render(IntPtr, Channel, System.Windows.Media.Visual, Int32, Int32, Double, Double, System.Windows.Media.Matrix, System.Windows.Rect)
   at System.Windows.Media.Imaging.BitmapVisualManager.Render(System.Windows.Media.Visual, System.Windows.Media.Matrix, System.Windows.Rect)
   at HelixToolkit.Wpf.Viewport3DHelper.RenderBitmap(System.Windows.Controls.Viewport3D, System.Windows.Media.Brush, Int32)
   at HelixToolkit.Wpf.BitmapExporter.Export(System.Windows.Controls.Viewport3D, System.IO.Stream)
   at Marlin3DprinterToolStlThumbnailProvider.StlThumbnailProvider+<>c__DisplayClass1_0.<GetThumbnailImage>b__0()
   at System.Threading.ExecutionContext.RunInternal(System.Threading.ExecutionContext, System.Threading.ContextCallback, System.Object, Boolean)
   at System.Threading.ExecutionContext.Run(System.Threading.ExecutionContext, System.Threading.ContextCallback, System.Object, Boolean)
   at System.Threading.ExecutionContext.Run(System.Threading.ExecutionContext, System.Threading.ContextCallback, System.Object)
   at System.Threading.ThreadHelper.ThreadStart()
```

```
Error Application Error
Faulting application name: explorer.exe, version: 10.0.16299.192, time stamp: 0x494a4408
Faulting module name: RGB9Rast.dll, version: 10.0.16299.15, time stamp: 0xb50dbd94
Exception code: 0xc0000005
Fault offset: 0x00000000000146d2
Faulting process id: 0x2d54
Faulting application start time: 0x01d38e379db58a7f
Faulting application path: C:\WINDOWS\explorer.exe
Faulting module path: C:\WINDOWS\SYSTEM32\RGB9Rast.dll
Report Id: 1646fb3c-29df-4ef0-83b8-790083bfb5f6
Faulting package full name: 
Faulting package-relative application ID: 
```

cabbagecreek, 2018-01-15:
> Ok, I think I found something though.... It is doing something with
> additional files inside the folder that are not .STL files. If I just put
> the STL files in a folder it renders them fine.
>
> If I add these additional files, it crashes. These are .mix files from
> Meshmixer and .b9L files from B9 Creator and 3mf files from Cura. These are
> all STL type files.
>
> https://1drv.ms/u/s!AkaMSvNd1E8DuGTDIQLgMm-d65At

Got the problem with "white" STL. There is no reason that the render should
be white..

I did not get a crash of WFE but it was stuck a long time on one of the STL.

I used Marli3DprinterToolSTLconfiguration to force a regeneration of all
thumbnail.. That helped but I got the white rendered STLs..

Maybe "Winter is comming" and this is some kind of "white walkers"

I will do some more debugging of this tomorrow..

I still think this is a much better version so I keep the update to
0.24-beta

The fix for the "white walkers" will be in v0.25-beta

Genae, 2018-01-16:
hey, sorry I answer a bit late but I was away last week. I tried your fixed version on a new Computer and still got the same errors. 
Here is the folder with the first view files:
https://1drv.ms/f/s!AtCtVPwlNMpogfphyUgyy2a6wzPdWw
Annis Hag seems to crash the Extension pretty reliably, pronably due to the big file size. Since @RonGeorge gets the same errors now those problems seem to be related.

Genae, 2018-01-16:
Ok, I found a fix for the USer device Registration Warning:
In the registry set
[HKEY_LOCAL_MACHINE \ SOFTWARE \ Microsoft \ Windows \ CurrentVersion \ WINEVT \ Channels \ Microsoft-Windows-User Device Registration / Admin]
set "Enabled" to"0".

While this fixes that warning the stl files keep crashing. But now with some interessting results:
[image]
[image]

cabbagecreek, 2018-01-16:
Looks like all the stl are Ok..

There is a totaly new version released (yesterday)

Run the Marlin3DprinterToolSTLconfiguration and UNREGISTER the existing STL
thumbnail. You should get 2 small messageboxes that confirms that its done.

Download the v0.24-beta

www.Marlin3DprinterTool.se/LatestVersion/Marlin3DprinterTool.msi

Uninstall existing M3DpT

To be sure that everything is removed, delete the C:\program
x86\marlin3dprintertool directory. It should be empty.

Install the new msi

Run M3DpTSTLconfiguration again and register the STL thumbnail Provider.

Run the Disk clean and check the thumbnail so all thumbnails will be
deleted. (start -> type "clean disk")

You may even have to reboot

This is my snap-shoot of the files.
[image: Infogad bild 1]

Genae, 2018-01-16:
ok, unregistered, uninstalled, rebootet, reinstalled, reregistered. It seemed to run pretty fine but than it crashed with the same error and the explorer restarted 4 times in a row.
[image]
It looks like the errors have been collected and then occured at the same time...

Genae, 2018-01-16:
if you need more files for testing: 
https://1drv.ms/f/s!AtCtVPwlNMpogfQSMY3ClHJacKcPew
(you can use 7zip to unzip them all at once)

Genae, 2018-01-16:
After some more testing the "collected errors" did not occure again, but the explorer keeps crashing. It does seem a lot more stable though.
I did also notice an white walker in there.

Some more system information:
Asus ROG G751J
Intel i7-4710HQ
Nvidia Geforce GTX 980M
24GB RAM
Windows 10 Education Insider Preview (The other Computer I tested on is not running Preview)
Version: 1709 (Fall Creators Update)
OS Build: 17025.1000

cabbagecreek, 2018-01-17:
Looks like "Close but no cigar"..
I have to do some deep debuging on this issue.
Are you willing to do testing with the dll creating a log for each file 
I create a "delete all logfiles"-program and a "find the log where WFE crashes"-program.
Then we can get a finetune of where the DLL actually crashes and what build of the DLL that is running.
I start working on a testplan and the supporting test-tools

RonGeorge, 2018-01-17:
I will, sure. Just send me the programs you want me to run.

cabbagecreek, 2018-01-17:
I have now spent 10 hours on this problem.
I got it alot more stable but still got some "White Walkers"
Now I can isolate the "whte Walkers" problem to just a few stl files... and they are rendered the same in the STLviewer so there is no differance here.. It must be within the model.

[Marlin3DprinterToolStlThumbnail.zip](https://github.com/cabbagecreek/Marlin3DprinterTool/files/1639469/Marlin3DprinterToolStlThumbnail.zip)

The attached zip contains the STL thumbnail provider..

To replace it you ned to unregister the STLtumbnailprovider.
then reboot
Now the STLthumbnailProvider can be replaced.

This DLL sends log to C:\TEMP so that directory nedds to be existing

This is for trace/debug only.. 

I think I am a bit closer to not crashing WFE. 
The problem left is
* Takes time
* Sometimes it needs to be refreched like change directory/ change detail-level to be able to view all thumbnailsComplex STL takes alot of time.. But it is a one time task

Genae, 2018-01-20:
I unregistered, then rebooted and then copied the dll to C:\Program Files (x86)\Marlin3DprinterTool. The i tried to register the provider again but got this error: 
[image]
Am I doing something wrong?

cabbagecreek, 2018-01-20:
try to install the v025-beta

www.Marlin3DprinterTool/LatestVersion/Marlin3DprinterTool.msi

I got this almost stable.. Still some whitewalkers but alot more stable

RonGeorge, 2018-01-20:
Ok, I can't get it to log. I tried to replace the .dll with your logging one and got the same error, I believe without reading it all, as above. So then I went and installed your v025-beta in the post above this, btw you forgot the .se on your URL, ;)

http://www.marlin3dprintertool.se/LatestVersion/Marlin3DprinterTool.msi

I uninstalled the old version
Installed the new version
Did "Install and Register STL" through interface

and now I can't get it to crash. I tried my crashing file, the other crashing file from above. I'm going to keep trying different files, but I saw you just posted a few minutes ago so I wanted to give you an update really quick.

Do you want me to do something to make it log? Want me to replace the v025 beta DLL with your logging one?

cabbagecreek, 2018-01-20:
the v0.25-beta is without logging.

I have done about 2 day debugging and got it more stable.. Still whitewalkers on some STL.
The same behavior on the STL Viewer. So there is something with the STL that creates white models..
Can the be an instruction within the STL that tells the STL that it should be white??
Must be that they are recording GoT and wants som more whitewalkers from the virtual world ;-)

RonGeorge, 2018-01-20:
I cannot get it to crash, I'll keep trying but it wont work.

Also, figured out the white walkers I think.

If the model was saved as STL ASCII format, it will be white... if it was saved as Binary, its the color you chose.. An example file in both formats attached.
[Desktop.zip](https://github.com/cabbagecreek/Marlin3DprinterTool/files/1648951/Desktop.zip)

RonGeorge, 2018-01-20:
Ok, I've tried both of my files. The first one I posted of the Orion Mecha, the one I just posted a few days ago "New Folder", I tried the folder that Genae posted, I also tried his Annis Hag.stl. None of them are crashing. I think you fixed it.

The only thing a little off, besides the White Walkers... is that the thumbnails are being rendered in my old, as in previous STL thumbnail choice (dark purple), and the viewer when I click the STL file shows it in the new color choice (
[image]
light purple).

Now, this is probably something wacky because I've installed and uninstalled it so many times, a registry setting is somewhere is probably zombied.

The only thing I would add would be for you to create this as its own separate program. If you did that and charge $2 for it, I think it would offset some of the time you have spent building ... literally the ONLY .stl thumbnail viewer around.

Nice work!!!

closed 2018-01-20 by RonGeorge

RonGeorge, 2018-01-20:
Just to add.. I also noticed something else. In Genae's folder of crashing files, I noticed that the problematic one renders a black background.
[image]

I also noticed that in my crashing folders there were files that had black backgrounds.
[image] *I've attached this set of stls for you*

[Grey_Knight_Terminator_Captain_-_72_hrs.zip](https://github.com/cabbagecreek/Marlin3DprinterTool/files/1648979/Grey_Knight_Terminator_Captain_-_72_hrs.zip)

Genae, 2018-01-21:
So I just installed the 0.25 version and it is FCK AMAZING ;) Great job! I renderd the folder with >800 stl files in < 2 min and with no crash. 
I personally dont care about the white walkers, since I am going to render all stls in white anyways, but here is what I noticed:
All white Files start with: "MESHMIXER-STL-BINARY-FORMAT" while the others are "Exported from Blender-2.71 (sub 0)". The blender files seem to be binary as well though.
As RonGeorge states, this binary format seems to be a part of the problem, but not all binary  formats are automatically broken.
About the instruction in the STL stating it top be white: This seems to be the case. When you look at the file in a Hex Editor you can see a lot of FF FF in a regular pattern:
[image]
Could be a white RGBA color.

Genae, 2018-01-21:
When looking at the meshmixer manual, it states that they do support multicolored stls:
"For Binary STL, we support face colors, but only one of the modes (there are two conventions and there is no way to tell which is which!). And we bake these face colors into vertex colors, so there is some blurring. "
http://www.mmmanual.com/export/
So I guess you are done, since you might not want to remove this "I can parse color from stl files feature" :P

cabbagecreek, 2018-01-21:
Wow!! Thanks alot Genae and RonGeorge.. Without you this should not be a success.
There is a reason why there is no stl thumbnail provider from some other software developer.
My conclusion for this is that it is difficult and that different software have different interpetations of what a STL is. The "whitewalkers" and the Genae investigations is marvelous.. I did not have a foggiest idea about this.
The "whitewalkers" is not a big problem because the "winter is comming".. In middle of Sweden its about -12C and in the nort its near -30C

The latest v0.25-beta is done alot with threads and should be alot faster. After almost 2 years of development I am as close as I can be with this.

As a gift to Genae and RonGeorge I will make a "divorce" from M3dPt and set the STL Thumbnail Provider as a separate program/Git but it will still be Open-Source. I dont belive in commersial Software. It should be free or as "beer-Ware". The separate program will be after 0.26-beta and I will do some new videos on youtube that cover this "new" program

cabbagecreek, 2018-01-21:
If you get the STL thumbnails in previous color.. Try running "Clean disk" and check the "Thumbnail" checkbox. This is the friendly way of deleting the STL taht will be recreated when you open a folder.
Important is to emty trashbin aswell.. 
Sometimes you have to do this twice or even more.
The thumbnails is so painful for WFE to create so it caches them and try every possibility to NOT recreate them. Thats why sometimes old versions pop-up along with new rendered files.
This is not a problem specific for STL. This is how Windows handles with thumbnails,

There is other more brutal ways. Just search for "Delete WFE Thumbnails". Take backup before doing some of the more brutal "thumbnail delete"

Genae, 2018-01-21:
One funny thing I noticed: The new version seems to be so much faster that the WFE does not try to cache all the images as hard as possible. When I open the Folder with all the stl files about 50% of them are cached (mostly the bigger files) while the rest is generated on the fly. It only takes a few seconds though so its fine. But if you are unable to leave the house due to a snow storm and do not know what to do, you might want to implement a custom image cache:
1. check if a thumbnail exists in %appdata%/stlThumbnail/filename_md5sum.png
2. if not create  a thumbnail and save it there.
this also has the benefit that you can replace the thumbnail images with custom images.
#feature#minor#niceToHave#open 
I am a happy man anyways so thanks for you effort!
Greetings from Munic, where the snow is not so sure yet wether it wants to stay or just flow away...