cabbagecreek/Marlin3DprinterTool #152: Marlin3DprinterTool and StlThumbnail installers conflict

glovax, 2018-02-14:
Johnny,
Unfortunately installation of Tools  and Thumbs goes into conflict. After I installed latest version of Tools and then Thumbs, Tools disappeared from disk. When I'm trying to install once again tools, I got a message - Newer version is already installed. They do not coexist. My workaround was to change folder name for Thumbs, uninstall it, and than Install Tools ;-) I know that its stupid but it works.
Tom

cabbagecreek, 2018-02-14:
Makes this an issue in Git

I have found this myself..

The solution is to have the 2 softwares installed as two totaly different
programs

Marlin3DprinterTool and STL thumbnails

This keeps the marlin3Dprintertool uncanged and STL thumbnails will have
duplicated dlls and config.

Regards
Johnny

cabbagecreek, 2018-04-21:
I have released v0.26-beta and this issue is solved

The STL viewer and M#DpT is joined in the same MSI (again)
There is an option to install STL and/or M3DpT

closed 2018-04-21 by cabbagecreek