cabbagecreek/Marlin3DprinterTool #161: Testing z-Probe

To1952P, 2018-04-11:
Measuring Z-Probe repeatability does not work. After the first calibration with the BLTouch no value is given and the process is stoped. The software is completely blocked and must be reset.

Version: 0.25.6607.18348
Marlin 1.1.8
Printer CR-10S
Using BLTouch

[image]

To1952P, 2018-04-13:
Already found the issue.
Z_MIN_PROBE_REPEATABILITY_TEST must be defined in Marlin
Line 726 Marlin 1.1.8 Configuration.h

cabbagecreek, 2018-04-16:
Perfect!!. During 3Dmeetup in Sweden last weekend I was doing alot if improvements.
I have totaly rewritten the Z-probe offset calculation.

cabbagecreek, 2018-04-21:
I have released v0.26-beta and this issue is solved

closed 2018-04-21 by cabbagecreek