cabbagecreek/Marlin3DprinterTool #163: Error on connection + delta support

marcottt, 2018-04-28:
Hi, i'm tryng this software, i wanna check if is useful for me to use and bed level my anycubic del plus. 
I've triend original bltouch but send back to amazon because processo to autolevel was not optimal (manuale bel level is oftne better) and the offset of the probe is too much... 

I tried marlin3dprintertool but when i try to connect i get "decryptation errpr: Dati non validi" (yes.. in italian)  after i get a "zylserialport.net - demo version by zylsoft"  message. 
Also i cannot find a way to setup a round bed... so is this software suitable for deltas's ?

i'm on marlin 1.1.8

cabbagecreek, 2018-05-03:
The decryption errpr is fixed.. ( ;-) for the spelling ... )
Also some issues with the bed level figures that was corrupted.
This fix also effect the syncronisation between the axis "kompass" and the visible values for current position.
This makes it possible to use the "kompass" jog control for Z during Z-Probe offset and makes that process a bit faster.

cabbagecreek, 2018-05-03:
Now the v0.27-beta is availablel

Closing this issue

closed 2018-05-03 by cabbagecreek