cabbagecreek/Marlin3DprinterTool #164: Configure Bed Issue

MikeLud, 2018-05-12:
I am having issues with Configure Bed. After homing all axis (G28) the front left corner would not probe. Below is a copy of the G code that was sent. I notice that the last G code (G30 S-1) does not look like a valid G code.

Thanks,
Mike

Send G28 Y
Receive X:0.00 Y:-7.00 Z:5.00 E:0.00 Count X:0 Y:-702 Z:2000
ok

Send G28 X
Receive X:-38.00 Y:-7.00 Z:5.00 E:0.00 Count X:-3813 Y:-702 Z:2000
ok

Send G28 Z
Receive X:135.00 Y:150.00 Z:10.00 E:0.00 Count X:13546 Y:15051 Z:4000
ok

Send G1 X30.0 Y50.0 Z10.00 F6000
Receive ok

Send G30 S-1

cabbagecreek, 2018-05-12:
The G30 will NOT execute if the probe is outside the bed.
So the setup of the bed is the probe position.. Not the nozzle position.
So redo the positions for the four corners and be sure that the probe is at the bed surface.

If the corners ar correct and the G30 will not execute the problem is in the formware and that the offest of the Z-probe is not correct. If the offset is wrong in the software the "think" that the probe is outside the bed and a G30 will crash the nozzle into the bed. Thats why the G30 dont execute.

A workaround  is to test the four corners about 50mm from each bededge. First corner like 50,50 and the secund at 50,150 on a 200x200 bedsurface. If this is working try to move closer to the edges , one by one. Then you get what offset that is wrong calculated.
This workaround asumes that your probe is closer than 50mm from the nozzle.

Use a marker pen on the bed surface to indicate where the probe meassures. If you have a PEI bed surface protect it with a pice of blue painters tape. This is just for calibration. remove the tape /marks when the positions are found and then do the bed level.

I belive some beds have point where the proximity sensor reads. I belive that there is a thin piece of metal in the bed that the proximity sensor feel. If the bed has such "spots" its important to aim at souch a point.

Proximity sensor on glasbeds can be improved with a razor blade between the glas and the bedheater. Same procedure here to aim at the prepared spots

If someone has a bed with this prepared spots I want to know. I specially want to know about the center position. I use the center position to have a point where I an sure that the Z home works and Thet the extruder is not in the way when adjusting. Maybe there is no senter pint in the bed with prepared spots and I have to add instructions in the software that the lower left corner should be used (0 , 0 +- zprobe offset) 

Comment if this instruction helped . I need to know the reason why this happend in the first case

MikeLud, 2018-05-13:
Thanks for the quick response. The issue was in my probe points settings. I added 5mm to the first x probe point and that fixed it.

Thanks,
Mike

closed 2018-05-13 by MikeLud