cabbagecreek/Marlin3DprinterTool #168: can not install marlin 3dprinter tool

bogaskilop, 2019-02-20:
i already uninstalled marlin 3d printer tool  deleted all registry entries  uninstalled also stl shell extension and still can not install  new version when i downloading newest version and run installer it says that newest version is already installed on pc  LOL

cabbagecreek, 2019-02-21:
I will try to do a zipfile for you without the msi installer.
You unzip the files intp C:/program (x86) /marlin3dprintertool

the disadvantage is that you dont get the munu items in the start menu

/J

Skickades från E-post för Windows 10

Från: bogaskilop
Skickat: Wednesday, February 20, 2019 5:42 PM
Till: cabbagecreek/Marlin3DprinterTool
Kopia: Subscribed
Ämne: [cabbagecreek/Marlin3DprinterTool] can not install marlin 3dprintertool (#168)

i already uninstalled marlin 3d printer tool deleted all registry entries uninstalled also stl shell extension and still can not install new version when i downloading newest version and run installer it says that newest version is already installed on pc LOL

gharryh, 2019-02-21:
That would be nice

bogaskilop, 2019-02-23:
ok i will do  menu for windows  that's not a problem but you can also add that files from your windows installation So i'm waiting for your zip file .

Thx

gharryh, 2019-02-23:
What files do you need in that zip file

bogaskilop, 2019-02-28:
Hi  gharryh   

When can i get those files ?  :) ??

gharryh, 2019-02-28:
Please tell me witch files you need??

bogaskilop, 2019-04-12:
Maybe all folder of installed marlin3d printer tool