cabbagecreek/Marlin3DprinterTool #170: Comunication always Crash

arnold-b, 2019-04-17:
Hello, 

i´m using the current Version and i tried on Windows 7 and Windows 10. 

If i´m using MarlinTool, after some time suddenly the comunication crash. Or if there is some Error on the Printer the communication also crash. 
After crash, i have to reconect the usb port and restart the application to work again. 

What can i check ?
thanks

cabbagecreek, 2019-04-18:
Hmm.. Sounds like an Issue..

What feature are you using when M3DpT crashes?

Is it possible that you can have ”Show Communication” window open. I want to se where it stalls

What version of Marlin FW

Regards

Skickades från E-post för Windows 10

Från: arnold-b
Skickat: Thursday, April 18, 2019 12:08 AM
Till: cabbagecreek/Marlin3DprinterTool
Kopia: Subscribed
Ämne: [cabbagecreek/Marlin3DprinterTool] Comunication always Crash (#170)

Hello,
i´m using the current Version and i tried on Windows 7 and Windows 10.
If i´m using MarlinTool, after some time suddenly the comunication crash. Or if there is some Error on the Printer the communication also crash.
After crash, i have to reconect the usb port and restart the application to work again.
What can i check ?
thanks

arnold-b, 2019-04-18:
I used Marlin 1.? before. (Current Bugfix 1.1.X) 

Now i´m using Bugfix2.0  and it works.

closed 2019-04-18 by arnold-b