cabbagecreek/Marlin3DprinterTool #174: Considering forking and updating some components of this software, but cannot build

evilC, 2020-04-14:
Any chance of some help building this solution?
All I am interested in at this point is the bed levelling tool.
I installed the software today, and used it to successfully level my bed, so it appears to still work with Marlin 2.0, but there were issues (Knob turn directions reversed)

CapnBob, 2020-04-16:
This works perfectly.  $10 :   
https://www.amazon.com/Audew-Digital-Tread-Depth-Gauge/dp/B078TCYPR7/ref=sr_1_10?ie=UTF8&qid=1532044794&sr=8-10&keywords=tire+tread+depth+gauge

[image]

https://www.thingiverse.com/thing:2991835

cabbagecreek, 2020-04-16:
What do you want to do? How can I help you? I will look at the latest commit and see if I have something that is not in Git that is needed /J Skickades från E-post för Windows 10 Från: CapnBobSkickat: Thursday, 16 April 2020 12:54Till: cabbagecreek/Marlin3DprinterToolKopia: SubscribedÄmne: Re: [cabbagecreek/Marlin3DprinterTool] Considering forking and updating some components of this software, but cannot build (#174) This works perfectly. $10 :https://www.amazon.com/Audew-Digital-Tread-Depth-Gauge/dp/B078TCYPR7/ref=sr_1_10?ie=UTF8&qid=1532044794&sr=8-10&keywords=tire+tread+depth+gaugehttps://www.thingiverse.com/thing:2991835

evilC, 2020-04-16:
Actually, I sat down yesterday and implemented what I wanted as a completely new app
https://github.com/evilC/MarlinProbeAssistedLeveler
It's only a console app for the moment, you need to edit values in the source to suit your setup, but it does work

cabbagecreek, 2020-04-17:
Lets do this together. You can ”promote” your probe together with M3DpT. Marvelous idea. Rhis was a sleeping Project and I have just woke it up. It would be nice to add some more features.Regards Skickades från E-post för Windows 10 Från: Clive GalwaySkickat: Thursday, 16 April 2020 20:04Till: cabbagecreek/Marlin3DprinterToolKopia: Johnny Lindén; CommentÄmne: Re: [cabbagecreek/Marlin3DprinterTool] Considering forking and updating some components of this software, but cannot build (#174) Actually, I sat down yesterday and implemented what I wanted as a completely new apphttps://github.com/evilC/MarlinProbeAssistedLevelerIt's only a console app for the moment, you need to edit values in the source to suit your setup, but it does work

evilC, 2020-04-17:
FYI your posts are coming through quite mangled on GitHub, looks like you are replying to the emails using an email client that is stuffing loads of HTML in the email
If you want to reach me for a chat, I have a Discord channel here: https://discord.gg/MmnhQYQ