cabbagecreek/Marlin3DprinterTool #179: not responding

ikn369330, 2021-09-27:
Can not get any response from  your Programme (Marlin3DprinterTool). I am using Windows 10 connected to a Bigtreetech GTR V1.0 board with Marlin 2.0.9.1 installed. When I select the correct board rate and Port I can do nothing else, if I click any tab I get a Program not responding error, I know its the right port and that connection is possible as I can connect and use Pronterface software no problem, I do not try and use the same programmes or ports at the same time. If I click reset on my TFT screen attached to my GTR board, I get an almost on and off connection as Marlin3DprinterTool shows my Marlin parameters but then no actions can be made, no movements, no PID tune, the programme allows me to select different tabs at this stage of quick on off, but nothing else
[image]
[image]
[image]

cabbagecreek, 2021-09-27:
Hello Ian!

It looks like the Marlin is configured to report Temp. M3dPt has connection with the board.
If you press “Show communication” you will se the communication live.
Look at this page ( https://marlinfw.org/docs/gcode/M155.html )
This is the command that creates the temp readings. 
In the “show communication” its possible to send the “M155 S0” that stops the temperature report.

Then there is an issue with Marlin 2.x that status report is not done like Marlin 1.x. Try a M501 = Read EEPROM (or the button )

That should “kick-start” the reading of initial parameters from startup of Marlin.

The reason for this wired behavior is that Temperature logging “blocks” the native GCODE commands that Marlin3DprinterTool is based on.

Try this and send me a mail of the result

Regards
Johnny Lindén

 

 <https://user-images.githubusercontent.com/16881000/134889713-fa58ad8d-294d-46d5-a101-db2b0d2d418f.jpg> 
 <https://user-images.githubusercontent.com/16881000/134889718-c9e45bc9-a634-4748-90cb-61ac27997bba.jpg> 
 <https://user-images.githubusercontent.com/16881000/134889721-9a2a99d8-0cbb-495b-837e-feff994b3e53.jpg> 

Detta e-postmeddelande har sökts igenom efter virus med antivirusprogram från Avast.
https://www.avast.com/antivirus

ikn369330, 2021-09-27:
HI There,
Thank you for getting back to me,
I am unable to use any commands as I can not establish a connection, however on my tft screen I was able to issue the M155 S0 command as for M501 it can not be done from using the button or issuing a command via 3dprinter tool, I can do this from the TFT or PC but to no effect

Regards

Ian Nicholas

<https://user-images.githubusercontent.com/16881000/134889713-fa58ad8d-294d-46d5-a101-db2b0d2d418f.jpg>
<https://user-images.githubusercontent.com/16881000/134889718-c9e45bc9-a634-4748-90cb-61ac27997bba.jpg>
<https://user-images.githubusercontent.com/16881000/134889721-9a2a99d8-0cbb-495b-837e-feff994b3e53.jpg>

https://www.avast.com/antivirus

cabbagecreek, 2021-09-28:
Hi,in a single gcode command

Have you done a Paypal donation. I can send the 5$ back,

I really do not know why it’s not working. It is plain gcode commands that are sent is sequences and parsed to find out what the 3D printer do and get data that is not available in a single gcode command.

Regards
Johnny

 

<https://user-images.githubusercontent.com/16881000/134889713-fa58ad8d-294d-46d5-a101-db2b0d2d418f.jpg>
<https://user-images.githubusercontent.com/16881000/134889718-c9e45bc9-a634-4748-90cb-61ac27997bba.jpg>
<https://user-images.githubusercontent.com/16881000/134889721-9a2a99d8-0cbb-495b-837e-feff994b3e53.jpg>

https://www.avast.com/antivirus

https://www.avast.com/antivirus