cabbagecreek/Marlin3DprinterTool #180: Some problem with serial communication

Regressor, 2021-12-10:
Hi. I got a ultimaker2 to repair. It got a hole inside X-driver and then 24V inside 5V line and all 5v related chips was burned. 

So I replaced drivers, cpu, usb and others and start to test it. I downloaded Marlin3DprinterTool, send some $ for beer, got license, pressed connect and then I realized that there is no communication with printer at all. There is can be single RX led flash once and then nothing (RX/TX leds on board are OFF). Show communication window is empty.

When I connects to serial port with putty or another tool and press keys I can see RX led flashing on printer pcb. I need this tool to test and calibrate all printer systems.
So what I'm doing wrong ?

I'm using windows 7.

cabbagecreek, 2021-12-11:
Hi,

There can be an issue with M503 command.

Try on the initial screen to press button M503 and see if the M503 = read
current settings will wake up the communication.
Also be sure that no other software has stolen the COMx port. If you run
Cura/Simplify3D it will establish communication and M3DpT will not be
allowed to use the COMx port.

Regards

Regressor, 2021-12-12:
Thank you. Yes. It was because of unfunctional printer firmware, I reflashed it and now Marlin3DprinterTool works good. I have some strange issue with xyz end switches (when I press x,y,z switches Marlin3DprinterTool allways lights minX + maxY + maxZ cirlces in endstop tab). I checked them using pronterface and they works ok - all axes stops with correct log message).

P.S. Is there any way to blowing separate check?